vhelin/wla-dx #340: [65816] -s flag parameter ignored for .sym file ?

hoit, 2020-12-08:
Hello,

From the [documentation](https://wla-dx.readthedocs.io/en/latest/wlaflags.html?highlight=wla%20options), the -s flag used with WLA do not create _sizeof_* definitions.

It seems to be ignored by wla-56816: all objects files have been built with the command line :
`F:\pvsneslib\devkitsnes\bin\wla-65816.exe -s -o myFile.obj myFile.asm`

Then after using wlalink with parameter to get the .sym file :
`F:\pvsneslib\devkitsnes\bin\wlalink -d -s -v linkfile test.sfc`

the .sym file still contains a lot of _sizeof__xxx inside.

I joined a sample here to reproduce the case, but some files has provided directly as .obj files for convenience reasons.
[tmp.zip](https://github.com/vhelin/wla-dx/files/5660841/tmp.zip)

Is it a bug ?

Also, i saw that in some cases, test.sym file become "corrupted" 
```
0006fff9 SECTIONSTART_.rodata
0006fff9 tccs_L.s3go.2
000710000 SECTIONEND_.rodata
```
the last line contains 1 zero more but for now i did not find a way to reproduce the code with a minimal sample. I can open a new issue for this one when i will have a sample to provide.

Thank you in advance

vhelin, 2020-12-10:
Hi!

It seems that WLALINK also created automatically some _sizeof_* definitions, not just WLA. So I added a flag (-D) to WLALINK to disable that... Give -s to WLA and -D to fully disable _sizeof_* definition creation...

That symbol file corruption I couldn't find in your test case... Please open a new issue for it if you manage to isolate the problem...

Thanks!

hoit, 2020-12-10:
Hi,

I just tried it and i confirm that it works :)

just to understand all details correctly with the error :
"TRY_PUT_LABEL: Definition "_sizeof_tccs_L.6" was defined more than once"

if this label was created with wlalink, i think if the same variable is declared in 2 different files, this is the reason that produce this error.
My case is solved and i do not use _sizeof_ variable but it can still produce duplicate. Maybe this name should be unique (by using the source file in the name for example?)

For the second issue, i am still at the same point and i did not find a way to reproduce it (seems to be random). I will open a new issue if i find it.

Thank you for this correction

closed 2020-12-10 by hoit