vhelin/wla-dx #341: .sym file sometimes corrupted on SECTIONEND

hoit, 2020-12-12:
Hi,

Sometimes, we saw that the .sym file generated by wla link is corrupted because one line in the file contains 1 zero more.

For instance :

```
0006fff9 SECTIONSTART_.rodata
0006fff9 tccs_L.s3go.2
000710000 SECTIONEND_.rodata
```

The last line should be 00071000 instead of 000710000

Without to be sure, the issue seems to appear only in "SECTIONEND_" lines, like you can see with other sample :

```
0009ffcf SECTIONSTART_.dmas0_text
0009ffcf dmaCopyCGram
000a10000 SECTIONEND_.dmas0_text
```

```
0008ffb3 SECTIONSTART_.pads0_text
0008ffb3 scanPads
000910000 SECTIONEND_.pads0_text
```

If i find a way to reproduce it i will update this issue but please tell me if you have any idea about the source of it.

Thank you in advance !

vhelin, 2021-03-14:
Hmm, I assume this happens only with 65816 code? wlalink/write.c:1433 (and similar) has

          fprintf(f, "%.4x%.4x %s\n", get_snes_pc_bank(l) >> 16, (int)l->address, l->name);

if l->address is 0x10000 then it outputs 5 digits, not 4 and you get one digit more than wanted... 

My questions are:
* should your first example be 00070000, not 000710000?
* should your second example be 000a0000, not 000a10000?
* should your third example be 00090000, not 000910000?

hoit, 2021-03-15:
I cannot affirm that it is only with 65816 because i did not used other assembler in wla dx.

I asked advise to Alekmaul (owner of pvsneslib) who is better but it is hard to provide you the response. In each case, there is one 0 more but not sure about the number 1.

I do not have any way to reproduce it each time, it is hard to test

vhelin, 2021-03-15:
I think this happens, because, in the first case, your section .rodata ends to the very last byte of the bank. After that byte we create a label SECTIONEND_.rodata, which overflows to the next bank as labels cannot be at the end of a byte, but at the start of a byte....

vhelin, 2021-03-15:
If you try with the latest sources, do you still get those extra zeroes in symbol files?

hoit, 2021-03-16:
pvsneslib has been updated with this new version of wla. Like the issue occurs only sometimes, i will take time to respond you definitively. I will keep you informed as soon as possible if i am sure the issue does not appear anymore.

Thank you for this change !

vhelin, 2021-03-16:
Argh, I didn't fix it properly! If you now take the latest sources it should be fixed... Sorry for the trouble!

vhelin, 2021-03-16:
I failed to see that get_snes_pc_bank() gives the correct bank even if address inside the bank is $10000...