vhelin/wla-dx #342: Bad memory write in macro_start_dxm

Drenn1, 2020-12-25:
My project has suddenly started getting segfaults. Investigation with valgrind gives me the following results on the latest commit 966f9c167e926be3a6da91d5eab13dee81f92031:

```
==552564== Command: wla-dx/binaries/wla-gb -o build/ages.o -DBUILD_VANILLA -D ROM_AGES ages.s
==552564==
==552564== Invalid write of size 8
==552564==    at 0x118390: macro_start_dxm (pass_1.c:353)
==552564==    by 0x122F06: directive_dbm_dwm_dlm (pass_1.c:2434)
==552564==    by 0x122F06: directive_dbm_dwm_dlm (pass_1.c:2398)
==552564==    by 0x1284AA: evaluate_token.part.0 (pass_1.c:1062)
==552564==    by 0x128A8A: evaluate_token (pass_1.c:1057)
==552564==    by 0x128A8A: pass_1 (pass_1.c:662)
==552564==    by 0x111594: main (main.c:186)
==552564==  Address 0x64b018c is 220 bytes inside a block of size 528 free'd
==552564==    at 0x483B9AB: free (vg_replace_malloc.c:538)
==552564==    by 0x11D3E3: undefine (pass_1.c:1309)
==552564==    by 0x11D3E3: directive_endm (pass_1.c:6903)
==552564==    by 0x1284AA: evaluate_token.part.0 (pass_1.c:1062)
==552564==    by 0x128A8A: evaluate_token (pass_1.c:1057)
==552564==    by 0x128A8A: pass_1 (pass_1.c:662)
==552564==    by 0x111594: main (main.c:186)
==552564==  Block was alloc'd at
==552564==    at 0x483CB65: calloc (vg_replace_malloc.c:760)
==552564==    by 0x1176BE: add_a_new_definition.part.0 (pass_1.c:1334)
==552564==    by 0x117929: add_a_new_definition (pass_1.c:1321)
==552564==    by 0x117929: redefine (pass_1.c:1282)
==552564==    by 0x117A4C: macro_start (pass_1.c:329)
==552564==    by 0x128F0D: pass_1 (pass_1.c:814)
==552564==    by 0x111594: main (main.c:186)
==552564==
==552564== Invalid write of size 4
==552564==    at 0x1183A1: macro_start_dxm (pass_1.c:360)
==552564==    by 0x122F06: directive_dbm_dwm_dlm (pass_1.c:2434)
==552564==    by 0x122F06: directive_dbm_dwm_dlm (pass_1.c:2398)
==552564==    by 0x1284AA: evaluate_token.part.0 (pass_1.c:1062)
==552564==    by 0x128A8A: evaluate_token (pass_1.c:1057)
==552564==    by 0x128A8A: pass_1 (pass_1.c:662)
==552564==    by 0x111594: main (main.c:186)
==552564==  Address 0x64b0194 is 228 bytes inside a block of size 528 free'd
==552564==    at 0x483B9AB: free (vg_replace_malloc.c:538)
==552564==    by 0x11D3E3: undefine (pass_1.c:1309)
==552564==    by 0x11D3E3: directive_endm (pass_1.c:6903)
==552564==    by 0x1284AA: evaluate_token.part.0 (pass_1.c:1062)
==552564==    by 0x128A8A: evaluate_token (pass_1.c:1057)
==552564==    by 0x128A8A: pass_1 (pass_1.c:662)
==552564==    by 0x111594: main (main.c:186)
==552564==  Block was alloc'd at
==552564==    at 0x483CB65: calloc (vg_replace_malloc.c:760)
==552564==    by 0x1176BE: add_a_new_definition.part.0 (pass_1.c:1334)
==552564==    by 0x117929: add_a_new_definition (pass_1.c:1321)
==552564==    by 0x117929: redefine (pass_1.c:1282)
==552564==    by 0x117A4C: macro_start (pass_1.c:329)
==552564==    by 0x128F0D: pass_1 (pass_1.c:814)
==552564==    by 0x111594: main (main.c:186)
==552564==
^C==552564==
==552564== Process terminating with default action of signal 2 (SIGINT)
==552564==    at 0x116D37: toupper (ctype.h:215)
==552564==    by 0x116D37: strcaselesscmp.part.0 (pass_1.c:175)
==552564==    by 0x11E861: strcaselesscmp (pass_1.c:171)
==552564==    by 0x11E861: parse_if_directive (pass_1.c:9712)
==552564==    by 0x125AB8: parse_directive (pass_1.c:8228)
==552564==    by 0x1284AA: evaluate_token.part.0 (pass_1.c:1062)
==552564==    by 0x128A8A: evaluate_token (pass_1.c:1057)
==552564==    by 0x128A8A: pass_1 (pass_1.c:662)
==552564==    by 0x111594: main (main.c:186)
==552564==
==552564== HEAP SUMMARY:
==552564==     in use at exit: 21,468,005 bytes in 35,328 blocks
==552564==   total heap usage: 95,981 allocs, 60,653 frees, 147,713,697 bytes allocated
==552564==
==552564== LEAK SUMMARY:
==552564==    definitely lost: 22,832 bytes in 2,202 blocks
==552564==    indirectly lost: 119,400 bytes in 244 blocks
==552564==      possibly lost: 0 bytes in 0 blocks
==552564==    still reachable: 21,325,773 bytes in 32,882 blocks
==552564==         suppressed: 0 bytes in 0 blocks
==552564== Rerun with --leak-check=full to see details of leaked memory
==552564==
==552564== For lists of detected and suppressed errors, rerun with: -s
==552564== ERROR SUMMARY: 2 errors from 2 contexts (suppressed: 0 from 0)
```

It is difficult to know exactly when the problem started, since the memory leak seems to appear and disappear. But it seems likely that this is the first commit in my project with the problem:
https://github.com/Drenn1/oracles-disasm/commit/3b1d0acd8c4056339aaed266ee68c1f6f27874fa

In this commit, I messed with one of my macros. The result now looks like this:

```
.macro m_BreakableTileData
	.if \3 > $f
	.fail
	.endif
	.if \4 > $f
	.fail
	.endif

	; Parameters 1-3 have their bits reversed.
	dbrev \1, \2
	revb \3
	.db (_out >> 4) | ((\4)<<4) ; "_out" is the output from the revb macro ("\3" with bits flipped).
	.db \5, \6
.endm
```

This uses by "revb" and "dbrev" macros for reversing bit order in bytes, which were only slightly changed in that commit:

```
.MACRO revb
	.define tmp \1
	.define tmp1
	.define tmp2
	.REPT 4 index tmpi
		.redefine tmp1 tmp&(1<<tmpi)
		.redefine tmp2 tmp&($80>>tmpi)
		.redefine tmp tmp&(($1<<tmpi)~$ff)
		.redefine tmp tmp&(($80>>tmpi)~$ff)
		.redefine tmp tmp | (tmp1<<(7-tmpi*2)) | (tmp2>>(7-tmpi*2))
	.ENDR
	.redefine _out tmp

	.undefine tmp
	.undefine tmp1
	.undefine tmp2
.ENDM

; Define a byte, reversing the bits
.MACRO dbrev
	.REPT NARGS
		.dbm revb \1
		.shift
	.ENDR
.ENDM
```

The nature of the bug makes this difficult to reproduce. Sometimes it even disappears and re-appears between some of my commits... I haven't been able to make an isolated test project.

Drenn1, 2020-12-25:
I should clarify that I've been using v9.11 without issue for some time, and it seems to have been a change it my project that triggered the bug, not a change in WLA-DX.

vhelin, 2021-01-17:
I haven't forgotten this, hopefully I have time to look at the issue next week. :)

vhelin, 2021-02-20:
Sorry for the delay, I'm looking into this right now. I think you've found something there...

vhelin, 2021-02-20:
If you try out the latest sources, are you still getting segfaults?

Drenn1, 2021-02-21:
This appears to have fixed the problem. Thanks!

closed 2021-02-21 by Drenn1