vhelin/wla-dx #347: You cannot use IFDEF inside a struct or enum declaration

jwrayth, 2021-01-07:
In my project I would like to include/exclude some struct and enum members based on if it's a debug or release configuration. 

>.ENUM $0000
    working dw
.IFDEF DEBUG
    not_working dw
.ENDIF
.ENDE

or

>.STRUCT TestStrict
    working dw
.IFDEF DEBUG
    not_working dw
.ENDIF
.ENDST

At present, the above generates the error;

> schema\wram.inc:41: DIRECTIVE_ERROR: Unexpected symbol "DEBUG" in .ENUM.
schema\wram.inc:41: ERROR: Couldn't parse "DEBUG".

It is possible to invert the setup and have different enum/struct declarations for debug or release, but this is cumbersome and more fragile as you need to maintain two declarations and layouts.

vhelin, 2021-01-07:
The parser in WLA is a bit crappy so your only hope at the moment is to create separate .ENUMs for DEBUG and RELEASE (?)... Perhaps one day someone can make the parser better, add a hack that fixes this issue, or code separate .IFDEF parser for .ENUM parser...

Drenn1, 2021-01-07:
I added this feature though. Which WLA-DX version are you using?

Drenn1, 2021-01-07:
See #150 and #152

jwrayth, 2021-01-07:
I was using v9. 7 but I have started moving to 9.11 today. I will report back once I have opportunity to test again.

vhelin, 2021-01-07:
It seems that I totally forgot that WLA supports this, and @Drenn1 seems to be right, this works...

.STRUCT testStruct
ok dw
.IFDEF DEBUG
not_ok dw
.ENDIF
at_last dw
.ENDST

... with the latest sources. I added a test for it.

vhelin, 2021-01-07:
I can confirm that with the latest sources .IFDEF inside .ENUM works also. I'll add a test for that as well...

jwrayth, 2021-01-07:
Thanks! I admit I should have tested against latest before opening the issue.

closed 2021-01-07 by jwrayth