vhelin/wla-dx #348: Macro \@ parsing has changed between 9.7b and 9.11, potentially broken

jwrayth, 2021-01-07:
Consider the following demonstration macro;

```
.MACRO exampleMacro
    bra partialLabel_\@+1
    nop
    partialLabel_\@:
    nop
.ENDM
```

In 9.7b, this builds as expected with the label being correctly resolved.

Starting in 9.11 you recieve the following error;

> FIX_REFERENCES: Reference to an unknown label "partialLabel_1"

If you insert a space between the \@ and the + operator, it assembles correctly. This leads me to believe there is a parsing error with the \@ arg when evaluating macros.

vhelin, 2021-01-07:
A new feature (support for small additions and subtractions with \@) in the parser turns partialLabel_\@+1 into partialLabel_1 if \@ is 0.

Any ideas how to keep the new feature, but make old code work? :)

jwrayth, 2021-01-10:
Since it changed an existing usage I think that's a good indicator that the parsing has ambiguity over what the intended result is.

So two solutions come to mind;

- Enforce strictness and have the new feature require brackets for scoping
  - partialLabel_(\@+1) for new behaviour
- Add warnings to the output similar to other compilers when an ambiguous choice has been made on your behalf
  - With the option to force warnings as errors for people like myself :P

vhelin, 2021-01-17:
Good solutions! I was supposed to work on this one this weekend, but had to do some extra work @ real life job... Well, perhaps next week?

vhelin, 2021-01-24:
I was unable to get partialLabel_(\\@+1) to work as the stack calculator ate those parentheses, but partialLabel{\\@+1} works now!

vhelin, 2021-01-24:
FYI, the commit description sucks, {LABEL_\@+1} doesn't work. What works is LABEL_{\@+1}...

jwrayth, 2021-01-25:
Thanks, that sounds useful! Is there any way to update the existing error message to assist people who may hit this? Or is that out of scope due to wider refactoring required?

vhelin, 2021-01-25:
If you now use

.MACRO exampleMacro
    bra partialLabel_\@+1
    nop
    partialLabel_\@:
    nop
.ENDM

... you don't get an error message any more, that bra should work as you want. If you want partialLabel_\\@+1 to turn into a label (partialLabel_1), then you are in trouble as that should be now written as partialLabel_{\@+1}... So all people who use v9.11's new "simple additions/subtractions with \\@" feature should be able to compile their code but the result is wrong...

closed 2021-05-25 by vhelin