vhelin/wla-dx #350: Add code style guidelines

lhsazevedo, 2021-01-10:
Add (or link to) a code style guidelines document to help newcomers properly style their code.
This could be also linked on CONTRIBUTING.md.

jwrayth, 2021-01-10:
This could be due to my recent change, and in fairness I had read the CONTRIBUTING.md, so it was my fault for not sanitizing my code to meet the guidelines beforehand. It would be useful to update it to be more explicit detail though as clarity is always clear (though I underline again my formatting was due to my own style and this repo being very different, and I should have addressed that before submission).

lhsazevedo, 2021-01-10:
Hey @jwrayth, I wasn't aware of your change.

I'm totally new to C and it's standands. Having clear guidelines could be helpful for newcomers like me, and for style consistency in the codebase.

vhelin, 2021-01-10:
Sorry for being strict about having all code written in the same style. I've been working on projects where everbody just used their own style so after a while people had been fixing bugs in other people's code it looked like a mess and was very hard to understand as the style could change after every 3-5 lines... :)

lhsazevedo, 2021-01-10:
I really like strict code style guidelines!

lhsazevedo, 2021-01-11:
Fixed by #353

closed 2021-01-11 by lhsazevedo