vhelin/wla-dx #356: Move active codebase out of root directory

lhsazevedo, 2021-01-11:
What do you think of moving the active codebase (wla-cpu, wlalink, byte-tester, etc) out of the root directory?
I think this way it's simpler to understand, navigate and maintain it. As well as keeping consistent with already separated folders (docs, examples, etc)

We can go even further and create a subdir for wla-cpu inside src/, and reorganize wla-cpu code in folders too (like decode/ and opcodes/):

```
📁 examples
📁 historical
📁 src
├ 📁 wlab
│ ├ 📄 main.c
│ └ ...
│
├ 📁 wlalink
│ ├ 📄 main.c
│ └ ...
│
└ 📁 wla-cpu
   ├ 📁 opcodes
   ├ 📁 decode
   ├ 📄 main.c
       ...
```

I've setup a branch so you can clone and test it: https://github.com/lhsazevedo/wla-dx/tree/move-to-src
It builds and passes all tests here at my Linux desktop. Don't know about Windows though.

cr1901, 2021-01-11:
It's a good idea, but I won't have time to test this on Windows or MSDOS (don't forget to update those Makefiles) for a bit.

In addition, changing the directory structure is going to make bisecting more difficult. But I suppose one could write a script for that to automate rebuilding against the old dir structure and new one.

lhsazevedo, 2021-01-11:
Sadly I can't test on other platforms, and it maybe be better to let more experienced people handle Makefiles as they are new to me to. I think this is more an idea/starting point than the real thing.

Oh, and what does bisecting means in this context?

cr1901, 2021-01-11:
By bisecting I mean [`git bisect`](https://www.metaltoad.com/blog/beginners-guide-git-bisect-process-elimination).

vhelin, 2021-01-11:
Good idea as the current folder structure is a bit messy. But like @cr1901 said, the historical makefiles/scripts need to be updated as well and work after the change. Personally I wouldn't do this unless I had nothing else to do as bug fixes & new features >>> reorganization... It'll be a pain in the behind to update those files in the historical folder... Let's keep this in mind unless someone really feels like doing it!

lhsazevedo, 2023-06-22:
Maybe we can consider this now that we have the makefile generator