vhelin/wla-dx #357: Certain operations on addresses no longer work

aexoden, 2021-01-12:
In a 65816 project I'm working on, I occasionally need to reference a 24-bit address in bank $7E, except with the $7E replaced with $00. (At least with LoROM on the SNES, the low $2000 bytes of RAM are mirrored to most banks, so this works fine.) I probably wouldn't write code this way, but I need to be able to match an existing binary which does this occasionally.

Here is a piece of test code that shows the problem:

```
.memorymap
	slotsize $8000
	defaultslot 1
	slot 0 $0000
	slot 1 $8000
.endme

.lorom

.rombanksize $8000
.rombanks 1

.bank 0 slot 1
.section "Bank0"
main:
	STA r_test.x.l
	STA r_test.x.l & $FFFF
	STA r_test.x.l,X
	STA r_test.x.l & $FFFF,X
.ends

.struct coordinate
	x db
	y db
.endst

.ramsection "LowRAM" bank $7E slot 0
	r_test instanceof coordinate 128
.ends
```

Prior to commit 5ab7533ad1d01848b27b5a32a37673c2d728fc11 this worked fine, but since then, I get the following error:

```
wla-65816 -o main.o main.s
wlalink -v -s linkfile result.rom
main.o: main.s:17: PARSE_STACK: Unresolved reference to "r_test.x".
make: *** [makefile:12: all] Error 1
```

If there's a better way to interpret the $7E0000 (in this case) as $000000 without doing the & $FFFF thing, I'd love to hear it, but that's what I had come up with at the time.

vhelin, 2021-01-12:
I cannot give an advice about that $7E/$00 issue, I really haven't written code for SNES, but about the error WLALINK gives you: These are the definitions you get in that LowRAM .RAMSECTION:

r_test
r_test.1
r_test.1.x
r_test.1.y
r_test.2
r_test.2.x
r_test.2.y
r_test.3
r_test.3.x
r_test.3.y
r_test.4
r_test.4.x
r_test.4.y
r_test.5
r_test.5.x
r_test.5.y
r_test.6
r_test.6.x
r_test.6.y
r_test.7
r_test.7.x
r_test.7.y
...

WLA used to create also r_test.x and r_test.y that were in fact the same as r_test.1.x and r_test.1.y, but as loose r_test.x and r_test.y made little sense we removed them. So you can fix the issue by writing

  STA r_test.1.x.l
  STA r_test.1.x.l & $FFFF
  STA r_test.1.x.l,X
  STA r_test.1.x.l & $FFFF,X

It seems that we forgot to update the manual about this change, I'll do it now...

aexoden, 2021-01-12:
It's actually kind of funny. I've been working on this for well over a year now. When I first started, those loose labels worked, and at some point, they stopped working, so I replaced them all. (I mainly used them on indexed reads, so it's an extra hint that the whole data structure's being accessed). At a later point, they started working again, so I started using them again. But now it seems they're gone again.

As long as I know what the intended behavior is. Replacing it seems to work with the first error, though I have a bunch of references to replace before I can be sure everything still builds. I'll close this for now and reopen it if I have more trouble.

closed 2021-01-12 by aexoden