vhelin/wla-dx #359: Building for CPM

StephanSchmidt, 2021-01-18:
I want to use wla-dx to develop a game for CP/M.

CP/M programs are z80 code and need to start at $0100.

When I use the following code, I can generate and run (hurray!) binaries on CP/M.
As wla-dx is designed for rom banks, the file is filled up with zeros.

How can I generate binaries that are only as large as the code and not filled up with 
zeros?

(The memorymap, rombankmap etc. are copied and pasted from the example and make no sense
for CP/M but work).

```
.MEMORYMAP
SLOTSIZE $4000
DEFAULTSLOT 0
SLOT 0 $0100
.ENDME

.ROMBANKMAP
BANKSTOTAL 2
BANKSIZE $4000
BANKS 2
.ENDRO

	.define bdos 5

        .orga 100h                                                                
start:  nop
	nop
	ld      de,msg                                                          
        ld      c,9                                                             
        call    bdos                                                            
	ret
msg:    .db    "Hello,world!",0dh,0ah,"$"      
```

vhelin, 2021-01-18:
You need to run WLALINK in program output mode (-b):

wlalink.exe -b -bS $100 -S linkfile output.prg

Also you need .MEMORYMAP and .ROMBANKMAP to be correct. .ROMBANKMAP in this case is just used as a placeholder for the assembler as internally it works with rombanks...

vhelin, 2021-01-18:
Ah, this .MEMORYMAP works better:

.MEMORYMAP
SLOTSIZE $4000
DEFAULTSLOT 0
SLOT 0 $0000
.ENDME

With this you get

$ ../binaries/wlalink.exe -b -S linkfile output.prg
Program start $100, end $119.

and in output.sym you'll see that label start is at $100...

[labels]
00:0100 start
00:010b msg

StephanSchmidt, 2021-01-18:
Excellent, thanks works! Also thanks for the fast turnaround.

closed 2021-01-19 by vhelin