vhelin/wla-dx #365: Prefix all global variables

lhsazevedo, 2021-01-24:
We need to add the `g_` prefix to all global variables, like we did [here](https://github.com/vhelin/wla-dx/pull/362).

Requested in https://github.com/vhelin/wla-dx/pull/362#issuecomment-766355851

vhelin, 2021-01-24:
This is something super nice that would make the code much, much better. Perhaps a modern IDE like Visual Studio could be used to rename the vars easily?

lhsazevedo, 2021-01-24:
Added a *very* incomplete progress list :)

> Perhaps a modern IDE like Visual Studio could be used to rename the vars easily?

Don't know about Visual Studio but I used VS Code for `i`, and it was a bit dumb. So I had to search for all local `i`s, rename then to `local_i`, rename all remaining `i`s to `g_source_pointer`, and name `local_i`s back to just `i`.

vhelin, 2021-01-24:
You did a bit more than was needed; you changed i's in e.g. WLALINK and WLAD to local_i, even though the global i problem was present in WLA - the assembler. But job got done and it seems to work - well done! :)

vhelin, 2021-01-24:
And yeah, VS Code ~= Visual Studio. Something more modern than the Emacs I use under Cygwin to develop WLA DX... I could ofcourse switch to Visual Studio as I'm using Windows 10 here... Hmm...

vhelin, 2021-01-24:
Also, if you take a look at WLA's main.c, these globals should also be renamed to contain the g_ prefix - the list is long:

char version_string[] = "$VER: wla-" WLA_NAME " 9.12a (9.1.2021)";
char wla_version[] = "9.12";

char *tmp_name = NULL;

extern struct incbin_file_data *incbin_file_data_first, *ifd_tmp;
extern struct file_name_info *file_name_info_first;
extern struct label_def *label_tmp, *labels;
extern struct map_t *global_unique_label_map;
extern struct macro_static *macros_first;
extern struct definition *tmp_def;
extern struct map_t *defines_map;
extern struct export_def *export_first, *export_last;
extern struct stack *stacks_first, *stacks_tmp, *stacks_last, *stacks_header_first, *stacks_header_last;
extern struct repeat_runtime *repeat_stack;
extern struct section_def *sections_first;
extern struct macro_runtime *macro_stack;
extern struct label_def *unknown_labels;
extern struct filepointer *filepointers;
extern struct map_t *namespace_map;
extern struct append_section *append_sections;
extern struct label_sizeof *label_sizeofs;
extern struct block_name *block_names;
extern struct stringmaptable *stringmaptables;
extern char mem_insert_action[MAX_NAME_LENGTH*3 + 1024];
extern char *unfolded_buffer, *label_stack[256];
extern char *include_in_tmp, *tmp_a;
extern char *rom_banks, *rom_banks_usage_table;
extern char *include_dir, *buffer, *full_name;
extern int include_in_tmp_size, tmp_a_size, *banks, *bankaddress;

int output_format = OUTPUT_NONE, verbose_mode = OFF, test_mode = OFF;
int extra_definitions = OFF, commandline_parsing = ON, makefile_rules = NO;
int listfile_data = NO, quiet = NO, use_incdir = NO, little_endian = YES;
int create_sizeof_definitions = YES;

char *final_name = NULL, *asm_name = NULL;

struct ext_include_collection ext_incdirs;

vhelin, 2021-01-24:
> Also, if you take a look at WLA's main.c, these globals should also be renamed to contain the g_ prefix - the list is long:
>
> char version_string[] = "$VER: wla-" WLA_NAME " 9.12a (9.1.2021)";
> char [...]

vhelin, 2021-01-24:
Sorry, pressed the wrong button... Too many beers after getting the laundry done. :)

vhelin, 2021-01-24:
I just renamed all global vars in WLALINK to start with "g_" using Visual Studio Community 2017 - It was pretty simple. No idea how to do that using Emacs. :) All tests pass so I assume it's working. Doing the same to the assembler (WLA) might take more time...

vhelin, 2021-02-12:
I think that this issue is now fixed. What do you people say, do you still see global vars without g_ prefix?

lhsazevedo, 2021-02-14:
Didn't see any non prefixed global :)

vhelin, 2021-02-14:
Ok, let's close it! :)

closed 2021-02-14 by vhelin