vhelin/wla-dx #366: Add simple Github Pages hosted homepage

lhsazevedo, 2021-01-24:
What do you think of having a *markety* homepage, hosted on [Github Pages](https://pages.github.com/)?

Some cool examples:
- [Moby Project](https://mobyproject.org/)
- [tiptap](https://tiptap.dev/)
- [cri-o](https://cri-o.io/)
- [Envoy Proxy](https://www.envoyproxy.io/)
- [Longhorn](https://longhorn.io)

vhelin, 2021-01-24:
During the past 20 years I've done four homepages:

- www.villehelin.com
- www.ttursas.com
- www.chillersoft.com
- www.kankibros.com (dead)

Three were done using Emacs and old HTML, one using some homepage creation tool back in 2011?

-> If someone else wants to do it, feel free to give it a shot. :)

vhelin, 2021-01-24:
PS. I'll still keep on updating http://www.villehelin.com/wla.html but if someone has the skills to do something fancy, that'd be nice to see!

lhsazevedo, 2021-01-24:
Candles of the Dead is really great!

vhelin, 2021-01-24:
Thanks :D

lhsazevedo, 2021-01-25:
Wrong button

lhsazevedo, 2021-02-06:
So, what does WLA DX mean?
There are any visual references that you think would be a good fit or inspiration?

I think I can suggest something :)

_PS: Would be awesome to know the history behind WLA DX!_

vhelin, 2021-02-06:
WLA DX comes from "Wzonka-Lad Assembler Deluxe"

- Wzonka-Lad - my old Gameboy emulator for Amiga
- Assembler, because it's an assembler, and the first versions supported only Gameboy GB-Z80
- Deluxe, because at the time I started to work on the assembler there were some games and apps that contained DX (Deluxe) in their name - it just was cool. :)

If I remember right, after getting Wzonka-Lad to work properly, I wanted to code games for Gameboy. I tried some assemblers back then, in 1998, but found them lacking. So instead of using assemblers I didn't like to make games for Gameboy my focus shifted to making a better macro assembler... :)

Never made a game for Gameboy, just graphics demos... Perhaps one day when WLA DX is good enough! :D

I have no idea of a visual reference for WLA DX's homepage. If you can suggest something, that'd be nice.

vhelin, 2023-01-31:
Great idea for a logo, love the WTF factory there for people who don't know what WLA DX stands for (plus the associations)... :)