vhelin/wla-dx #367: Possibility for a strict instruction mode?

jwrayth, 2021-01-29:
I'm targeting the 65816 and I have continually had issues were WLADX infers the wrong sizing for an instruction/operand. From reading past issues, I understand that this is a difficult problem to solve based on the current implementation between the assembler and the linker.

What I would like to suggest is; Would it be feasible to introduce a "strict sizing" mode that requires the programmer to **always** explicitly declare the size suffix for the instruction/operand and disable all inference? Personally, I would find this much more useful than trying to decorate my code with sizing hint directives.

I'm not sure where to start looking at a feature like this in the codebase, so I wanted to raise it as an issue first to get some expert opinion.

vhelin, 2021-01-29:
The place to implement this feature might be in decode_*.c files, I think, as there source code is turned into WLA's internal symbols.

Enhancement idea: Currently decode_*.c files are difficult to edit, so perhaps encapsulating the code inside to a function would make life easier?

jwrayth, 2023-02-22:
Sure thing! So in my 65816 project I'm not using directives like `.16BIT` to allow instructions to infer size, I'm using explicit operand sizing on my instructions.

For example, instead of 
```
.16BIT
lda #$80
```

I would use

```
lda.w #$80
```

I was looking for a mode to disable any inferred sizing, and force the user to explicitly declare the size of the instruction (when applicable). This is so that I can receive errors when the operand sizing has been omitted.

vhelin, 2023-02-22:
Just a sidenote (mainly for @jeffythedragonslayer): Instead of giving

lda.w #$80

which is a bit nasty if you think it from the point of a MC68000 programmer, you can also give

lda #$80.w