vhelin/wla-dx #368: Packed ASCII Support

StephanSchmidt, 2021-01-30:
Make .ASC support packed ASCII where characters are encoded into 6bits. 

This saves 25% of storage - with the downside of only uppercase characters.

Before printing a string one would need to decode this to 8bits.

vhelin, 2021-01-30:
Taking  lowercase letters away from ASCII -> doesn't really fit into 6 bits. Something else also has to go... Any links to specifications of this 6-bit version of ASCII? Is there a 6-bit ASCII standard in old games?

I found this: https://en.wikipedia.org/wiki/Six-bit_character_code - SixBit ASCII?

StephanSchmidt, 2021-01-30:
You are right, more has to go. And there are different encodings, e.g.

https://programmingpraxis.com/2014/05/06/packed-ascii/

which might be DEC SIXBIT from your link.

Kroc, 2021-01-30:
This is too specific for the assembler; compression should be done via a python script or the like and .INCluded. (There are literally infinite number of encoding schemes people might want). There's also the file-parsing macro that runs a macro for each byte in a file. 

What would be useful, in a generic sense, that you could use to achieve packed ASCII, is the ability to output less than 8-bits at a time (e.g. `.BITS ...`?)

StephanSchmidt, 2021-01-30:
It's much more difficult to include a packed string file than packed ASCII in the source file with address labels for each string.

But yes if this is not included I'd used some python script to generate the packed ASCII and generate an include file with the labels.

vhelin, 2021-02-01:
WLA already contains some directives that make life easier so adding one more should be ok, as long as it's really useful. :)

.BITSMAP 4 "0123456789" ; takes the number of bits the mapping uses and a mappings array?
.BITS "668275" ; maps the string using the .BITSMAP into 4 bits per character?

Quite doable, if this is ok and usable?

StephanSchmidt, 2021-02-01:
Something that way would be nice, indeed.

I currently read about how Infocom did text storage for their text adventures.

They use something they have called ZSCCI which encodes lower case characters with 5 bits, and upper case characters with more bits, and punctuation with even more.

This can lead to up to a 50% reduction in memory usage for texts (but is more complicated to encode/decode), though the open source Z-machine virtual machines should have code.

https://www.inform-fiction.org/zmachine/standards/z1point0/sect03.html

vhelin, 2021-02-01:
Sounds like some form of ZIP/deflate compression (Huffman?). I wrote one deflate/inflate tool back in the day, but cannot remember the details any more. Faster libraries do the work for me these days. :)

Anyway, if people think .BITSMAP and .BITS would be useful, I can add them to WLA...

Kroc, 2021-02-01:
I'm personally against this for two reasons:

1. It's too specific for one use-case. What happens if you want to also barrel-shift the bits, use a XOR encoding and so on and so forth. Not least, if you're using symbols from a lookup table rather than raw ASCII. There will always be another use-case where the current implementation isn't right. This is why I prefer the functionality of sub-byte ouput over bit-mapping

2. Bit-packed ASCII is not effective compression and what's more is slow for 8-bit systems to decompress -- a byte-compression scheme such as byte-pair encoding and the new FSST method (https://github.com/cwida/fsst) will completely out do any bit-packing method in speed and size. If you want to compress text, bit-packing is a bad method. LZ-methods often use Huffman trees which are, again, best served by sub-byte output and not bit-mapping

vhelin, 2021-02-01:
That FSST looks nice, I'm going to read that PDF tomorrow, but I'm thinking that could such .BITSMAP/.BITS still be useful for beginners? People who want to save some bits and not want to dig deeper into rabbit holes like data compression? Or are people in general in two camps when it comes to 8-bit programming: "don't compress at all" or "do it really well"?

Also, what's the difference between (in 1.) "sub-byte ouput" (outputs less than 8-bits) and "bit-mapping" (outputs less than 8-bits)?

I have no idea how much people use real hardcore data compression in their 8-bit projects versus using something like this "drop the unused 2 bits" method. I could add support for this feature, but if it's just StephanSchmidt's who is using it, it'd be, well, a little bit wasted effort. :)

vhelin, 2021-03-22:
I'm currently thinking of doing (thanks @Kroc for the ideas!)

.ARRAYDEF (define an array) - create a mapping table
.ARRAYIN (put a value into an array) - initialize the mapping table
.ARRAYOUT (read a value from array to a define) - do the mapping

plus

.BITS (define n bits)

With these directives one could do packed ASCII and plenty more

vhelin, 2021-03-28:
Ok, if you take the latest sources you can do bit packing like this:

        // define an array for mapping ASCII values into
        // less bits
        .ARRAYDEFINE NAME MapArray SIZE 4

        .ARRAYIN NAME MapArray INDEX 'A' VALUE %0000
        .ARRAYIN NAME MapArray INDEX 'B' VALUE %0001
        .ARRAYIN NAME MapArray INDEX 'C' VALUE %0010
        .ARRAYIN NAME MapArray INDEX 'D' VALUE %0011
        .ARRAYIN NAME MapArray INDEX 'E' VALUE %0100
        .ARRAYIN NAME MapArray INDEX 'F' VALUE %0101
        .ARRAYIN NAME MapArray INDEX 'G' VALUE %0110
        .ARRAYIN NAME MapArray INDEX  0  VALUE %1111

        .MACRO MapInto4Bits
        .ARRAYOUT NAME MapArray INDEX \1 DEFINITION MAPPING
        .BITS 4 MAPPING
        .IF \1 == 0
            .ENDBITS
        .ENDIF
        .ENDM

Takes quite a few lines, but it should work. :)

Kroc, 2021-03-28:
That's great at having the flexibility to do both!

vhelin, 2021-03-28:
I think I should also add support for this:

        .ARRAYIN NAME MapArray INDEX 'A' VALUES %0000, %0001, %0010

Starting index would be 'A' - you could use that instead of writing

        .ARRAYIN NAME MapArray INDEX 'A' VALUE %0000
        .ARRAYIN NAME MapArray INDEX 'B' VALUE %0001
        .ARRAYIN NAME MapArray INDEX 'C' VALUE %0010

vhelin, 2021-03-28:
Ok, the example just got smaller:

        // define an array for mapping ASCII values into
        // less bits
        .ARRAYDEFINE NAME MapArray SIZE 4

        .ARRAYIN NAME MapArray INDEX 'A' VALUES %0000, %0001, %0010, \
            %0011, %0100, %0101, %0110
        .ARRAYIN NAME MapArray INDEX  0  VALUE %1111

        .MACRO MapInto4Bits
        .ARRAYOUT NAME MapArray INDEX \1 DEFINITION MAPPING
        .BITS 4 MAPPING
        .IF \1 == 0
            .ENDBITS
        .ENDIF
        .ENDM

        .FILTER MapInto4Bits "BAGED", 0

vhelin, 2021-04-18:
Apparently .ARRAYIN/.ARRAYOUT etc. are working as there are no bug reports about them? :)

closed 2021-04-18 by vhelin