vhelin/wla-dx #375: Make g_inz and g_ind local vars when possible

vhelin, 2021-02-07:
In the assembler global vars g_inz and g_ind are used all over the place instead of local vars. Replace all uses of them both with local vars when possible.

lhsazevedo, 2021-02-07:
We can make a list of good candidates to local conversion and post here so we can label this as a good-first-issue, what do you think?

vhelin, 2021-02-07:
Anything that helps. :) But the issue is in pass_1.c and decode_*.c files - just searching for g_ind/g_inz there will give you lots of opportunities for fixes. The decode_*.c files might have to be done at the same time as they are linked to the loop in pass_1.c...

vhelin, 2022-10-02:
decode.c now needs this treatment

lhsazevedo, 2022-10-10:
I would like to do that, can you assign me to this issue?

closed 2023-04-29 by vhelin