vhelin/wla-dx #376: Add more tests

vhelin, 2021-02-07:
Currently the "examples" folder contains 57 tiny/messy projects that are used as tests. Some folders contain a "testsfile" that is used with byte_tester.exe to confirm that WLA+WLALINK have managed to output correct bytes. Not all folders contain this verification file.

- Add more folders to examples that use random features of WLA/WLALINK
- Add more "testsfile"s to example folders

I assume that this issue is never solved as there never can be enough tests. :) 

PS. Rename that "examples" folder to "tests" and create some proper examples/project templates that people can use as starting points for their new projects...

cr1901, 2021-02-07:
@vhelin Last year I mulled on extending `minunit.h` for WLA DX. `ctest` would be fine for all the platforms supporting `cmake`, but I don't want the non `cmake` platforms to get left behind. 

So I got sidetracked over how to create a wrapper binary for a test suite and... well, I didn't get far thanks to the pandemic derailing everything (this was April 2020) :P.

I think `minunit.h` and this [SO answer](https://stackoverflow.com/questions/5609692/building-tests-with-cmake-while-not-using-ctest) is a good starting point to unite a test suite binary from CMake and vintage land :D.

vhelin, 2021-02-07:
It'd be awesome to get the tests to run on e.g., Amiga and MSDOS! Currently they run only on platforms where /bin/sh is working... (Linux, Cygwin...)... That'd be a great leap!

Even getting the tests to work on Windows platforms would be nice. :)

lhsazevedo, 2021-02-10:
I think we should really focus on this. A good test suite can avoid regressions when adding a new feature or refactoring.

Unfortunately I don't know enough C and Cmake to even try to create a unit test suite and unit tests :/ And I didn't even mention Amiga and MSDOS

But I can write some feature tests though using the current runner or another runner that you setup.

vhelin, 2021-02-12:
Making the tests run on Amiga / MSDOS would just require proper scripts per platform. I think the byte_tester.exe the tests use compiles just fine everywhere. I can try to take a look at how to run the tests on classic Amiga OS 2.04/3.0... And yeah, you need to make makefiles or something like them, for the tests, for Amiga / MSDOS... Not a quick thing to do.

More tests using the current runner - yes please. :) If you ask me, this should be priority one. As long as we don't have anything better this is the way to go. And if we one day get a better way of running the tests, the current tests should be included/converted as most of them are quite useful.

Drenn1, 2021-03-13:
We also need a way to test for things that should fail (ie. ".db 1 % % % % % % 5" currently assembles)