vhelin/wla-dx #387: Generate listfile output on macros

Drenn1, 2021-03-07:
It seems that listfiles don't generate any output on macros. For example, use of a macro like this appears to produce no listfile output, but if possible, it should show the two bytes defined when the macro is used:

```
.MACRO definetwobytes
  .db \1, \2
.ENDM

definetwobytes 1 2
```

I have no idea how difficult this would be to add, but it would be very useful if it could be added.

vhelin, 2021-03-07:
I think it would mean that the current list file generator is at least partially rewritten as currently it doesn't support multiple entries for the same source file & linenumber...

Rewriting the whole list file generator would be nice as currently it's too complex for what it does. :)

vhelin, 2021-03-07:
Just fixed the listfile generator a little, and went through the code - it sorts the outputted data by linenumbers fed to WLA's internal symbol stream. It does support for multiple entries per line, but to make .MACRO/.REPT output to work with listfiles we might need to drop the sorting and assume that everything the listfile generator has read is sorted already (it should be?)...

vhelin, 2021-05-30:
Ok, .MACRO calls generate now data in listfile. The data is put where the .MACRO is called from as it's possible to call a .MACRO from multiple different locations.

vhelin, 2021-06-22:
I assume the commit https://github.com/vhelin/wla-dx/commit/77a758f9d4ded35cd9409b19ccd9880c9a862843 fixed this issue, so I'll close this issue. If it's not working in some cases, please open a new issue.

closed 2021-06-22 by vhelin