vhelin/wla-dx #388: .IF statement improvements

Drenn1, 2021-03-11:
I find the current capabilities of IF statements rather limiting.

For starters, there is no .ELIF directive (#67). This isn't necessarily hard to add, the only real problem is that we have multiple ".IF" variants. So if we had ".ELIF", we'd also need - at a minimum - ".ELIFDEF", ".ELIFDEFM", ".ELIFEXISTS", ".ELIFNDEF", ".ELIFNDEFM". This could be done, but it's not the most elegant solution.

C compilers don't do this. Instead, "#IF" and "#ELIF" are flexible enough to do everything by themselves; ie. you can do "#elif defined X" instead of "#elifdef X".

I could potentially envision a built-in function in the vein of issue #309 called "defined" that could be used like this:

```
.IF defined(X)
[...]
.ELIF defined(Y)
```

(I have said before that my preferred syntax for "functions" in WLA would involve brackets like that, although this wasn't agreed upon.)

But I may be getting ahead of myself. Perhaps the greatest improvement could be made by giving .IF statements a full-on expression parser. This could be done in a similar way as ".DEFINE" statements work, perhaps even sharing the same arithmetic parser - with the condition evaluating to true if the result is nonzero. That way we could check for multiple conditions at once:
```
.IF defined(A) && defined(B)
[...]
.ENDIF
```

Instead of doing ugly stuff like this:
```
.IFDEF A
.IFDEF B
[...]
.ENDIF
.ENDIF
```

Obviously "&&" doesn't exist yet even for .DEFINE's, but it could be easily added as a new boolean operator. Or we could use "&" which would work almost as well, although not quite the same.

I could potentially implement some of this myself, but most of these things require some discussion, which I'm hoping to stimulate with this issue.

vhelin, 2021-03-11:
The

.IF defined(A) && !defined(B) && exists("hello.bin") && D > E
.ELIF defined(C)
.ELSE
.ENDIF

route sounds the best to my ear. If you could help the project out by adding support for this, then please do! :) The ideal solution would do all this and be compatible with old .IFs.

I can give this a couple of hours this weekend, but I don't think that's enough... All help is appreciated. :)

One thing that I have in my mind just now is that could we enhance the stack calculator (in stack.c) to parse these "calculations"? Or would it be better to make a similar calculator for conditions?

...

I'm also happy if we just get support for .ELIF - anything helps.

Drenn1, 2021-03-11:
Reusing the stack calculator is exactly what I had in mind, so .IF statements would work the same as .DEFINE's or .DB's. Conversely, this would mean that the boolean operators we add (&&, ||, >=, etc) will also be available in those contexts. Existing ".IF" statements should work exactly the same if we simply add the necessary boolean operators to the stack parser, so there shouldn't be any problem compatibility-wise.

That would be the first step, which I might see about doing myself tomorrow. After that we can implement the "functions". I'm less sure about how to handle that elegantly although I guess it shouldn't be too difficult.

Drenn1, 2021-03-15:
Just a little update on this, I did get started on it and I'll keep working on it over the week. The moment I touched the stack calculator I felt the need to start reorganizing it a bit... and write some tests for it to make sure it doesn't break... one thing leads to another I guess, haha.

vhelin, 2021-04-18:
I'm currently working on a C like compiler that uses a couple of files from WLA DX (mainly parse.c and stack.c), and there I had added support for logical ANDs and ORs etc. that made parsing complex conditions possible. So I thought that I'd backport my mods to WLA DX. It wasn't that easy as WLA DX did string comparisons, too, that had to be implemented, and I hope everything still works, but now we are just missing .ELIFs...

Drenn1, 2021-04-18:
ok cool, sorry I never got around to implementing this like I said! Look great.

cr1901, 2021-04-18:
Adding `.ELIF` should also close #67 :D.

vhelin, 2021-04-18:
> ok cool, sorry I never got around to implementing this like I said! Look great.

No worries. :)

vhelin, 2021-04-18:
> Adding `.ELIF` should also close #67 :D.

My question is: Are the new features working well? :) It'd be great to hear some feedback as making .ELIF and complex conditions to work required quite a lot of modifications...

I wrote some tests, but I guess I should write even more...

vhelin, 2021-04-18:
> Adding `.ELIF` should also close #67 :D.

This took what 5.5 years to happen? :) Sorry about that...

cr1901, 2021-04-18:
@vhelin I mean, I could've implemented it in the meantime, but didn't. Right, I still owe you an email explaining what happened...

vhelin, 2021-04-18:
No worries, this is a hobby project. After I released v9.5 in 2008 WLA DX was dead for seven years. I'm pretty happy where WLA DX is today. :)

vhelin, 2021-04-20:
Ok, I wrote more tests for this new .IF/.ELIF/.ELSE/.ENDIF, found a bug and fixed that. This issue should now be fixed, but if you find more bugs please reopen this issue.

closed 2021-04-20 by vhelin