vhelin/wla-dx #389: Documentation is little broken?

vhelin, 2021-03-14:
1. If I go to https://wla-dx.readthedocs.io/en/latest/ is see that the title in the tab is "<no title> - wla-dx 9.12a documentation" (also on Google search), and on the web page is reads: "Docs » <no title>"

2. The documentation doesn't compile any more under my CygWin (latest binaries). All I get is this:

[ 99%] Generating simple text documentation
Traceback (most recent call last):
  File "/usr/bin/sphinx-build", line 7, in <module>
    from sphinx.cmd.build import main
ModuleNotFoundError: No module named 'sphinx'
make[2]: *** [doc/CMakeFiles/doc-stext.dir/build.make:102: binaries/doc/wla-dx.txt] Error 1
make[1]: *** [CMakeFiles/Makefile2:1196: doc/CMakeFiles/doc-stext.dir/all] Error 2
make: *** [Makefile:172: all] Error 2

lhsazevedo, 2021-08-03:
Just built locally, title Ok. Maybe it's something related to ReadTheDocs?

[image]

vhelin, 2021-08-04:
Just visited readthedocs.io, now it looks like this - still broken

[image]

I guess ReadTheDocs is somehow not working properly, perhaps they use some special version of sphinx, but this is still an (tiny) issue. Perhaps we need to modify the document sources somehow differently to get ReadTheDocs to work? Perhaps someone knows what's going on? :)

sverx, 2022-08-17:
I can't see any problem. This is how it appears to me (Vivaldi, on Linux)

[image]

vhelin, 2022-08-17:
I'm also seeing this at the moment

[image]

I guess when I some time ago went and did something to the read-the-docs settings accidentally solved this issue (made the robots to use a newer version of some tools) as it seems to be ok now.

closed 2022-08-17 by vhelin