vhelin/wla-dx #390: Add support for Super FX

vhelin, 2021-03-17:
The instruction set looks like it could be easily added to WLA:

  https://en.wikibooks.org/wiki/Super_NES_Programming/Super_FX_tutorial

but there are some things that I don't currently understand:

  * "add R1" -> is it %01010001 or %00010101 ? (which nybble is the top nybble?)
  * "lm (Rn, xx)" ("Load word data from RAM, using 16 bits") -> but is xx here 8-bit?
  * "lms (Rn, yy)" ("Load word data from RAM, short address") -> what is short address, yy?

I'm sure plenty of more questions come up if one starts to implement support for Super FX... Or are there any better documents out there?

cr1901, 2021-03-18:
@vhelin You'll want this: https://archive.org/details/SNESDevManual/book2

vhelin, 2021-03-18:
Holy moly! Thanks for the link, I'll dig into these books this weekend!

cr1901, 2021-03-18:
Just FYI: SuperFX should support all directives the 65816 backend does- it has the same banking architecture. You could do a much better job than I adding it, but it would be fun to test :P.

vhelin, 2021-03-18:
Perhaps we should add support also for DSP-1? I see there are chips called DSP-2, DSP-3 and DSP-4 as well. No docs about them? :)

vhelin, 2021-03-18:
Thanks for the link, now my next weekends are booked. :D And yes, I'm taking the Super FX support implementation to myself...

cr1901, 2021-03-19:
Re: the DSP chips, @Ramsis-SNES is correct- SNES games access the DSP chip via an API on the 65816 side. Additionally, the DSP is, well, a DSP. It has only "4" instructions `OP`, `RT`, `JP`, and `LDI`, but each instruction embeds multiple instructions (VLIW).

This is a very long shot, but in principle, a DSP1/2, etc assembler may be valuable for experimentation. The firmware for the DSPs is dumped, and while I won't link them, they are not hard to find. WLA-DX's overlay could be used to compare revisions or try bug fixes/enhancements.

There's two issues I can think of w/ supporting the DSPs:

1. You wouldn't want to only support "4" instructions, you'd want to support the "microinstructions" inside the "macroinstructions", and then once per line (or some other terminator, like a semicolon?) combine all the microinsns into a macroinsn to determine the opcode. See the "An assembly language OP" paragraph on page 13 of this datasheet: https://datasheetspdf.com/pdf-file/526540/NECElectronics/UPD77C25/1
2. It's a Harvard Arch... data pointer 0 != code pointer 0.

vhelin, 2021-03-30:
Is the Super FX support working as you guys would like it to work? Any changes that needs to be done?

vhelin, 2021-03-30:
Hmm, I see some additional information in https://problemkaputt.de/fullsnes.htm#snescartgsunprogrammablerisccpuakasuperfxmariochip10games - I'll fix BGE/BLT etc tomorrow...

vhelin, 2021-04-30:
It's been a month and no bug reports about the new Super FX support, so I'll close this...

closed 2021-04-30 by vhelin