vhelin/wla-dx #392: Add support for DSP-1

vhelin, 2021-03-18:
... the chip used in some SNES games.

vhelin, 2021-03-30:
Ok, it seems that you don't write programs for DSP-1 etc. but you just give them a command and data via registers, and they'll execute the command (calculation), and return the result.

cr1901, 2021-03-31:
@vhelin Indeed, that's correct. However, it may still be useful (if low priority) to write an overlay for the existing DSP-1 program ROM: https://github.com/vhelin/wla-dx/issues/390#issuecomment-802453374

vhelin, 2021-03-31:
True that, my previous comment was just a memo for me about what's going on... :)

KungFuFurby, 2023-07-10:
I think what you're looking for is support for the ASM from the NEC µPD77C25. The DSP series of chips are hard-coded programs embedded into the chip, and I think the NEC µPD77C25 is the basis.

I'll refer you to to this documentation for more details: https://www.ersinelektronik.com/class/INNOVAEditor/assets/Datasheets/D77C25.pdf

vhelin, 2026-05-25:
Coming back to this, if DSP series contain just hard coded programs, then I guess adding NEC µPD77C25 support to WLA DX will make the lives of SNES developers any better? I mean is there any point in NEC µPD77C25 support in WLA DX?

vhelin, 2026-08-16:
I think I'll close this issue as I invented it without knowing better. If a need for this comes from outside I'll reconsider this.

closed 2026-08-16 by vhelin