vhelin/wla-dx #396: 65c816 : variable not initialised without using -x flag ?

hoit, 2021-03-21:
Hi,

In the current version of wla (took from master branch), we saw that variables are not initialized in all cases when using -x flag or not during building phase.

Here is the c file where the issue occurs with **myGlobalVar** variable:

```
#include <snes.h>
extern char snesfont;
u8 myGlobalVar = 6; 

int main(void) {
	consoleInit();
   
	consoleInitText(0, 0, &snesfont);
	setMode(BG_MODE1,0);  bgSetDisable(1);  bgSetDisable(2);

	consoleDrawText(10,10,"Hello World !");

	setScreenOn();  
	
	while(1) {
		WaitForVBlank();
	}
	return 0;
}
```

The .c file is translated to asm and i have compared with and without -x flag with WinMerge tool. If we ignore the difference of variable name (because the name is randomly generated), the asm file are the same.

But when we link the files, some DISCARD instructions appears and **myGlobalVar** is not initialized without -x flag :

_without -x flag :_

```
Building: wla-65816 -s -o hello_world.obj hello_world.asm
...
Linking ... hello_world.sfc
wlalink -d -D -s -v -L /z/Repository/pvsneslib/pvsneslib/lib linkfile hello_world.sfc
DISCARD: hello_world.obj: hello_world.asm: Section "ram.data" was discarded.
DISCARD: hello_world.obj: hello_world.asm: Section ".data" was discarded.
DISCARD: hello_world.obj: hello_world.asm: Section ".bss" was discarded.
```

_with -x flag_

_Building with -x flag: wla-65816 -s -x -o hello_world.obj hello_world.asm
...
Linking ... hello_world.sfc
wlalink -d -D -s -v -L /z/Repository/pvsneslib/pvsneslib/lib linkfile hello_world.sfc_

I added in attachment the output files generated in both cases, hope it will help to see the issue :

[logs.zip](https://github.com/vhelin/wla-dx/files/6178167/logs.zip)

Is it a bug ?

Thank you in advance

vhelin, 2021-03-23:
It kinda looks like a bug as WLA's -x shouldn't affect section discarding in any way. But I see that you are giving -d (discard unreferenced sections) to WLALINK in both cases and as I don't see any references in the code to e.g., "ram.data" so it should be discarded without any side effects...

Are you sure your C compiler (?) generates working assembly code? Or do you have other .asm files in the project that I cannot see here that make the references that I cannot see here?

vhelin, 2021-03-23:
Perhaps it'd be best if you asked also in the support forums (or GitHub Discussions? Issues?) of the C compiler project you are using, I don't even know which compiler you are using. :)

hoit, 2021-03-23:
I confirm that the C compiler generate working assembly code. Before upgrading to the latest version of wla, we were using wla 9.4 (i know it is very old). The behaviour was different and the "bug" was not existing.

I did not provided other .asm files because it is all sources of PVSneslib tool, the library i use.
The .c sources are converted to asm with tcc816 tool (based on Tiny C compiler) but it is not maintained anymore but it do not seems to be the root cause, because the asm file generated is the same with or without -x flag.

The sample i provided use 4 other files generated before : crt0_snes.obj, libc.obj, libm.obj and libtcc.obj.
Some .obj files are generated with these rules :

```
C_SRC	:=	malloc.c background.c console.c dma.c sound.c sprite.c video.c
libc.obj: libc.asm libc_c.asm
libc_c.asm: libc_c.c $(C_SRC) sm_spc.asm 
libc_c.c: $(C_SRC)
libm.obj: libm.asm libm_c.asm
```

and the ram.data sections are declared in libc_c.asm (line 12 835) and in libm_c.asm (line 2 185) :

[libc files.zip](https://github.com/vhelin/wla-dx/files/6193329/libc.files.zip)

vhelin, 2021-03-23:
Have to go to sleep, I'll return to this tomorrow. But does crt0_snes.obj contain any reference to myGlobalVar? Are crt0_snes.obj, libc.obj, libm.obj and libtcc.obj assembled with the same flags?

hoit, 2021-03-24:
Sorry, my previous sample was not complete to understand the case and i mixed different things.

To illustrate the issue, i modified the .c source file to add an action on myGlobalVar :

```
#include <snes.h>

extern char snesfont;

u8 myGlobalVar = 6; 

int main(void) {
	consoleInit();
    
	consoleInitText(0, 0, &snesfont);

	setMode(BG_MODE1,0);  bgSetDisable(1);  bgSetDisable(2);

	consoleDrawText(10,10,"Hello World !");
	
	myGlobalVar += 3;

	setScreenOn();  
	
	while(1) {
		WaitForVBlank();
	}
	return 0;
}

```

myGlobalVar should be equal to 9 but it is not the case.

I added a folder with the minimal code to reproduce the issue, you just have to build hello_world with and without -x flag to see the difference then linking it (the 2 command lines are provided in the directory with sources) :

[logs.zip](https://github.com/vhelin/wla-dx/files/6196930/logs.zip)

When it is built without -x flag, these sections are discarded :

DISCARD: hello_world.obj: hello_world.asm: Section ".data" was discarded.
DISCARD: hello_world.obj: hello_world.asm: Section ".bss" was discarded.

and it is not the case when the flag -x is used.

vhelin, 2021-03-24:
Thanks for the code that reproduces the issue! I'll check it out this evening after I've quitted working for today...

vhelin, 2021-03-24:
It seems that crt0_snes.obj has references to  SECTIONEND_.data and  SECTIONEND_.bss and  SECTIONSTART_.data and  SECTIONSTART_.bss so if you are not using -x flag for WLA then those labels are not generated and there are no references to .data and .bss sections and they can be discarded.

I'm not yet sure why you don't get an error for a missing reference, I'll look into that next, but it seems your compiler needs those labels and thus it needs the -x flag.

vhelin, 2021-03-24:
Any chance of you uploading crt0_snes.asm so we could take a look? :)

vhelin, 2021-03-24:
Lines 251 and 260 of crt0_snes.asm are of interest here. :) It seems to calculate the size of the sections...

----------------------------------------------------------------------
stack_item: label              : SECTIONEND_.data
stack_item: label              : SECTIONSTART_.data
stack_item: operator           : -
id: 3 file: crt0_snes.obj line: 251 type: 1 bank: 0 position: 1 section_status: 1 section: 3000004 result: 1520/$5f0 (ROM) 1520/$5f0 (RAM)
----------------------------------------------------------------------
stack_item: label              : SECTIONEND_.bss
stack_item: label              : SECTIONSTART_.bss
stack_item: operator           : -
stack_item: value              : 65534.000000/$fffe (RAM) 65534.000000/$fffe (ROM)
stack_item: operator           : &
stack_item: value              : 2.000000/$2 (RAM) 2.000000/$2 (ROM)
stack_item: operator           : +
id: 4 file: crt0_snes.obj line: 260 type: 1 bank: 0 position: 1 section_status: 1 section: 3000004 result: 2118/$846 (ROM) 2118/$846 (RAM)
----------------------------------------------------------------------

vhelin, 2021-03-24:
It seems that libc.obj has also such sections, thus linking doesn't fail:

FINDING LABEL SECTIONEND_.data... GOT IT! libc.obj: libc_c.asm:12849
FINDING LABEL SECTIONSTART_.data... GOT IT! libc.obj: libc_c.asm:12842
FINDING LABEL SECTIONEND_.bss... GOT IT! libc.obj: libc_c.asm:12882
FINDING LABEL SECTIONSTART_.bss... GOT IT! libc.obj: libc_c.asm:12852

Renaming those sections in libc_c.asm to something else might be a good start... ?

hoit, 2021-03-24:
You completely lost me :D

I think by creating this sample with -x flag to illustrate the issue, i added more bugs to other library files (like crt0_snes.obj or libc.obj).

Sources should be always built like that :

```
#---------------------------------------------------------------------------------
# for libc.asm file only, we need to use -x flag of wla dx to create labels SECTIONSTART_[section name]
# and SECTIONEND_[sectionname] at the beginning and end of each section.
#---------------------------------------------------------------------------------
%.obj: %.asm
	@echo Doing obj files ... $(fname)
	@CURRENTFILE=$<; \
		if [ $$CURRENTFILE = libc.asm ]; then \
		echo "Building with -x flag: $(AS) -s -x -o $@ $<"; \
		$(AS) -s -x -o $@ $<; \
		cp $@ tmp/$@; \
	else \
		echo "Building without -x flag: $(AS) -s -o $@ $<"; \
		$(AS) -s -o $@ $<; \
		cp $@ tmp/$@; \
	fi
```

The -x flag was useful only for libc.asm file.

So hello_world.obj is built without -x flag and the discard of sections during linking seems to produce the issue.

In this archive i uploaded, i added crt0_snes.asm file and all files are built with the hereabove rule.

[minimal_code_to_reproduce.zip](https://github.com/vhelin/wla-dx/files/6201088/minimal_code_to_reproduce.zip)

Tomorrow, i will synchronise with Alekmaul (developer of PVsneslib) to be sure about the case and provide you more details if necessary, because it is no more clear for me anymore

vhelin, 2021-03-25:
If you can use -x only with libc.asm, but if you use it with hello_world.asm and only then will the result work, then well that's a problem with the compiler. Which is something that I cannot fix for you.

``
   ; copy .data section to RAM
    ldx #0
-   lda.l SECTIONSTART_.data,x
    sta.l SECTIONSTART_ram.data,x
    inx
    inx
    cpx #(SECTIONEND_.data-SECTIONSTART_.data)
    bcc -

    ; set data bank register to bss section
    pea $7e7e
    plb
    plb

    ; clear .bss section
    ldx #(((SECTIONEND_.bss-SECTIONSTART_.bss) & $fffe) + 2)
    beq +
-   dex
    dex
    stz.w $2000, x
    bne -
+
``

This does tell us that something is going on with .data and .bss sections, but what - I cannot tell as I know nothing about the compiler you are using to generate this assembly code. And if userspace files like hello_world.asm contain the .data and .bss sections like crt0_snes.asm, should crt0_snes.asm process them all, not just what's inside crt0_snes.asm...

In general, according to this, crt0(_snes.asm) in this case, possibly, might do the initialization of all variables:

  https://en.wikipedia.org/wiki/Crt0

This case is not clear to me as well. :D

hoit, 2021-03-28:
> And if userspace files like hello_world.asm contain the .data and .bss sections like crt0_snes.asm, should crt0_snes.asm process them all, not just what's inside crt0_snes.asm...

I do not really understand how works asm but do you mean that section name need to be unique ?
because .bss and .ram.data exists in hello_world.asm and also in some files of the pvsneslib library.

In the sample provided with my previous message, the hello_world.asm file comes from this .c file :
```
#include <snes.h>

extern char snesfont;

u8 myGlobalVar = 6; 

int main(void) {
	consoleInit();
    
	consoleInitText(0, 0, &snesfont);

	setMode(BG_MODE1,0);  bgSetDisable(1);  bgSetDisable(2);

	consoleDrawText(10,10,"Hello World !");
	
	myGlobalVar += 3;

	setScreenOn();  
	
	while(1) {
		WaitForVBlank();
	}
	return 0;
}
```

in hellow_world.asm, i see this part :

```
...

tsa
clc
adc #8
tas
lda.w #0
sep #$20
lda.l myGlobalVar + 0
rep #$20
clc
adc.w #3
sep #$20
sta.l myGlobalVar + 0
rep #$20

...

.ramsection "ram.data" bank $7f slot 3
myGlobalVar dsb 1
.ends
.section ".data" superfree
.db $6
.ends
```

If i am not wrong, the `.db $6` in .data section is this instruction : `u16 myGlobalVar = 6; `
the `adc.w #3` is for `myGlobalVar += 3;`

The .sym file generated display : 007f05f0 myGlobalVar

but in no$sns, when i go to this adress, i have :

007F:05F0   0DFAFF00FF00FF00FF00FF00FF00FF00

If i understand it correctly, 0DFA is equal to 3578 but the variable should be equal to 9, that is why i think it related to wla.

If it is not the case, i will investigate in the previous version we used (wla 9.4) which maybe worked differently for our case.

vhelin, 2021-03-28:
> I do not really understand how works asm but do you mean that section name need to be unique ?
> because .bss and .ram.data exists in hello_world.asm and also in some files of the pvsneslib library.

What I mean is that if you have a .data and .ram.data sections in each object file for global variables, but crt0_snes only processes the first one its (references are) linked to, then only global variables in one file work and the rest are ignored.

I'm assuming that crt0_snes works like crt0's in general and initializes global variables.

Here you should take a deeper look into how the compiler works and why it's not processing all .data sections... Or is it coded so that it should process all .data sections, but for some reason with a new WLA DX it just processes one?

> in hellow_world.asm, i see this part :
>
> ```
> ...
>
> tsa
> clc
> adc #8
> tas
> lda.w #0
> sep #$20
> lda.l myGlobalVar + 0
> rep #$20
> clc
> adc.w #3
> sep #$20
> sta.l myGlobalVar + 0
> rep #$20
>
> ...
>
> .ramsection "ram.data" [...]

Looks like it!

> The .sym file generated display : 007f05f0 myGlobalVar
>
> but in no$sns, when i go to this adress, i have :
>
> 007F:05F0 0DFAFF00FF00FF00FF00FF00FF00FF00
>
> If i understand it correctly, 0DFA is equal to [...]

Yeah, well, if the compiler worked with WLA DX v9.4 and doesn't work with WLA DX v9.12+, then something has changed in WLA DX or the compiler itself relied on a bug or some "feature" that's been fixed...

I'm sorry I cannot help that much in this case as I haven't coded anything in 65816 asm ever, and I don't know the compiler you are using...

hoit, 2021-04-01:
Hi,

Sorry to bother you again with this issue, but we are still working on it and did not find any way to solve it.
After checking crt0_snes.asm, this file seems to be correct and do not seems to be the cause of the issue.

We did other tests we 2 files with global variables and the behaviour is strange, the files have been build with -x option.
Here is the data section (you will find it in the attachment)

[src.zip](https://github.com/vhelin/wla-dx/files/6243552/src.zip)

```
.ramsection "ram.data" bank $7f slot 3

fnt dsb 12
titi dsb 1

.ends

.section ".data" superfree

.dw snesfont + 0, :snesfont
.dw snesfont + 0
.dw :snesfont
.dw snesfont + 0
.dw :snesfont
.db $6
.ends

```
and

```
.ramsection "ram.data" bank $7f slot 3

toto dsb 1

.ends

.section ".data" superfree

.db $19
.ends

```
Asm files and .sym are equivalent to this .c code :

```
#include <snes.h>

extern char snesfont;

const u8 *fnt[]=
{
    &snesfont,&snesfont,&snesfont
};

char titi=6;

extern void updtoto(void);
extern char toto;

int main(void) {
	consoleInit();

    consoleInitText(0, 0, fnt[0]);
    
    titi=titi+2;
    updtoto();
    toto=toto+3;

	setMode(BG_MODE1,0);  bgSetDisable(1);  bgSetDisable(2);

	consoleDrawText(10,10,"Hello World !");
	consoleDrawText(6,14,"WELCOME TO PVSNESLIB");
	consoleDrawText(3,18,"HTTPS://WWW.PORTABLEDEV.COM");

	setScreenOn();  
	
	while(1) {
		WaitForVBlank();
	}
	return 0;
}
```

and 

```
#include <snes.h>

char toto=25;

void updtoto(void) {
   
    toto=toto+2;
    
}
```

in the .sym file, "toto" variable is not in ram.data section : 
```

007f0000 SECTIONSTART_ram.data
007f0000 fnt
007f000c titi
007f000d SECTIONEND_ram.data
0000b9c4 SECTIONSTART_.data
0000b9d1 SECTIONEND_.data
```

and you can see it with no$sns :

[image]

**titi** variable equal to 8, it is correct (6 + 2) and **toto** equal to f3 because the 2nd init part has been ignored but toto variable is on the adresse 007f000d

If you have any clue or explanations, we will be very happy :)

vhelin, 2021-04-02:
Hmm, my clue is to find the place in the compiler's asm files where toto is copied to RAM and initialized.

hoit, 2021-04-02:
thank you, your clue is clear. But i have a conception issue in this case because the code which copy data to ram is this one :

```
    ; copy .data section to RAM
    ldx #0
-   lda.l SECTIONSTART_.data,x
    sta.l SECTIONSTART_ram.data,x
    inx
    inx
    cpx #(SECTIONEND_.data-SECTIONSTART_.data)
    bcc -
```

But wla rename ramsection "ram.data" bank $7f slot 3 to a unique name in each file.
So for the 1st file, it become SECTIONSTART_ram.data, for the 2nd: SECTIONSTART_ram.data_1 and so on...

toto is visible in .sym file but under an other section, that is why it does not work

```
007f000d SECTIONSTART_ram.data_1
007f000d toto
007f000e SECTIONEND_ram.data_1
```

Is it possible to add variable to the "default" section instead of creating a unique section name ?

Otherwise, i need to see how the code can be modified to do initialisation of all global variables in crt0_snes.asm (a kind of joker like SECTIONSTART_ram.data* !)

vhelin, 2021-04-02:
It seems that as we are using hashmap these days (this was not coded by me) to store labels, we rename duplicate global definitions/labels in WLALINK so that they can be stored in a hashmap. Which is pretty stupid, if you ask me. :) I guess we didn't use hashmaps in WLA DX v9.4...

It seems that label/definition renaming is not necessary, perhaps? Can you try with the attached wlalink.zip?

[wlalink.zip](https://github.com/vhelin/wla-dx/files/6250857/wlalink.zip)

vhelin, 2021-04-02:
> But wla rename `ramsection "ram.data" bank $7f slot 3 `to a unique name in each file.
> So for the 1st file, it become `SECTIONSTART_ram.data`, for the 2nd: `SECTIONSTART_ram.data_1` and so on... [...]

Just to be sure, WLALINK doesn't rename the sections, but it renames those global SECTIONEND_ram.data and SECTIONSTART_ram.data labels as they are duplicates on the global level.

In the wlalink.zip I've removed duplicate checking and label renaming... If they work for you then the problem is solved.

> Is it possible to add variable to the "default" section instead of creating a unique section name ?
> 
> Otherwise, i need to see how the code can be modified to do initialisation of all global variables in crt0_snes.asm (a kind of joker like SECTIONSTART_ram.data* !)

My question is:

Does this code handle all 1-100+ (one per file) data + ram.data sections, or just one?

; copy .data section to RAM
    ldx #0
-   lda.l SECTIONSTART_.data,x
    sta.l SECTIONSTART_ram.data,x
    inx
    inx
    cpx #(SECTIONEND_.data-SECTIONSTART_.data)
    bcc -

Is it in the same file where sections data and ram.data are defined,  so it should work locally? If not, then you have initialization code for one section and multiples of such sections...

I think one working solution would be that every .asm file initialized its own data + ram.data sections, and crt0_snes.asm would call the initializer for every .asm file in the project. This might need that every data and ram.data sections had unique name, the initializer function had its own unique name per file, and your compiler would generate the calls to be put in crt0_snes.asm... ?

vhelin, 2021-04-02:
And yes, you can merge .SECTIONs and .RAMSECTIONs using APPENDTO keyword:

https://wla-dx.readthedocs.io/en/latest/asmdiv.html#ramsection-vars-bank-0-slot-1-align-256-offset-32

.RAMSECTION "ram.data.1" APPENDTO "ram.data"
label2    DB
.ENDS

.RAMSECTION "ram.data.2" APPENDTO "ram.data"
label3    DB
.ENDS

...

hoit, 2021-04-05:
Hi vhelin,

Thank you for these informations !

> It seems that label/definition renaming is not necessary, perhaps? Can you try with the attached wlalink.zip?
> 
> EDIT: The changed files vs latest sources are: main.c write.c analyze.c
> 
> [wlalink.zip](https://github.com/vhelin/wla-dx/files/6250857/wlalink.zip)

I tried the version in attachment and i confirm that i do not have anymore unique sections name but as i have differents sections, i think our code will not work.

> My question is:
>
> Does this code handle all 1-100+ (one per file) data + ram.data sections, or just one?
>
> ```
> ; copy .data section to RAM
>     ldx #0
> -   lda.l SECTIONSTART_.data,x
>     sta.l [...]

I am not good in asm but i think this code will only handle one section. Each file have its own data and ram.data section that is why we probably need to change the code too.
Your solution could help us ("every .asm file initialized its own data + ram.data sections, and crt0_snes.asm would call the initializer for every .asm file in the project") and i need to discuss with PVSneslib owner to know which solution choose and how to implement it

Thank you for the APPENDTO instruction, i tested it quickly and it still generated me 2 different sections, i need to do more tests to understand where it comes from.

I will keep you informed soon :)

vhelin, 2021-04-05:
> I am not good in asm but i think this code will only handle one section. Each file have its own data and ram.data section that is why we probably need to change the code too.
Your solution could help us ("every .asm file initialized its own data + ram.data sections, and crt0_snes.asm would call the initializer for every .asm file in the project") and i need to discuss with PVSneslib owner to know which solution choose and how to implement it

I hope you guys can do this! With unique section names, that'd be a long lasting solution...

vhelin, 2021-04-05:
Sorry, pressed the wrong button... :)

hoit, 2021-04-09:
After some investigation on your propositon ("every .asm file initialized its own data + ram.data sections, and crt0_snes.asm would call the initializer for every .asm file in the project"), we did not find a way to implement it. I am checking all other way to have something which work with the latest wla version (by using APPENDTO and KEEP instructions) and i found strange cases :

The KEEP instruction should avoid to discard my section but it seems not work :

```
.include "hdr.asm"
.accu 16
.index 16
.16bit
.ramsection "mainram.data" bank $7f slot 3 KEEP
.ends

```

```
wla-65816 -s -x -o mainsection.obj mainsection.asm
DISCARD: mainsection.asm: Discarding an empty section "mainram.data".
```

Does it comes from the way i use it ?

The -x flag which generate section_start and section_end seems to be executed after the check to discard sections or not. In my case, i use directly the final section name (section_start_ram.data) but the section is discarded.
The "discard" code seems to be executed file by file, so if i use APPENDTO instruction to add the content of ram.data section to a section in another file, i think it will not work ?

Thank you

vhelin, 2021-04-09:
If you now use the latest sources empty sections that are marked with KEEP are no longer discarded. This was a separate check from "is this section referenced" check in WLALINK and done in WLA...

vhelin, 2021-04-09:
I just tested, you can use APPENDTO with .SECTIONs/.RAMSECTIONs that are in separate files.

Currently appending sections in WLALINK doesn't touch the SECTIONSTART_* and SECTIONEND_* labels that are generated by WLA, so they are out of sync. I guess I need to update those after the sections are appended?

hoit, 2021-04-11:
Thanks a lot for these improvements, we progressed on the code upgrade and it is better than before but we still have some issues.

I think there is a tiny bug when using the APPENDTO instruction. When i have multiple files with `.SECTION ".data" APPENDTO "glob.data" `it produce me the warning :

_MERGE_SECTIONS: Multiple source sections called ".data" found, using the last one for append. Please rename one of the sections for a more predictable outcome._

In case of using APPENDTO, i think the section name to test should not be `.data` but the one where you add data: `glob.data`

vhelin, 2021-04-11:
You get the error, because APPENDTO-data is just sectionA-to-sectionB string pairs in the object files, separated from the sections themselves. So if your project has multiple ".data" sections and they are using APPENDTO, then the linker is confused.

I think I could try to make it so that when the APPENDTO-pairs are read from the object file I link quickly the source section to this APPENDTO-pair and the linker doesn't have to look for the source.

Give me an hour or two... :)

vhelin, 2021-04-11:
Ok, if you now get the latest sources your APPENDTO-issue should be gone. Note that file formats have changed so you'll need to reassemble everything.

hoit, 2021-04-11:
Very reactiv ;) I just tried it and i get few errors:

[minimal_code_to_reproduce.zip](https://github.com/vhelin/wla-dx/files/6292115/minimal_code_to_reproduce.zip)

I think this one is new (it produce on different sections) :
```

Z:\Repository\minimal_code_to_reproduce\minimal_code_to_reproduce>Z:\Repository\pvsneslib\devkitsnes\bin\wlalink -d -D -s -v linkfile hello_world.sfc
hello_world.obj: hello_world.asm:92: TRY_PUT_LABEL: Label "SECTIONSTART_ram.data" was defined more than once.
hello_world.obj: hello_world.asm:94: TRY_PUT_LABEL: Label "SECTIONEND_ram.data" was defined more than once.

```

and this one too

```
crt0_snes.obj: crt0_snes.asm:267: COMPUTE_STACK: The passed on SLOT changed from $2 to $0. This might have no effect, but just to let you know. Please check that the result of this calculation is correct.
_fprintf_snes_label(): Label's address inside a bank is 10490 > $FFFF! Please submit a bug report!
_fprintf_snes_label(): Label's address inside a bank is 11a41 > $FFFF! Please submit a bug report!
_fprintf_snes_label(): Label's address inside a bank is 11c62 > $FFFF! Please submit a bug report!
_fprintf_snes_label(): Label's address inside a bank is 11c92 > $FFFF! Please submit a bug report!
_fprintf_snes_label(): Label's address inside a bank is 11cc6 > $FFFF! Please submit a bug report!
_fprintf_snes_label(): Label's address inside a bank is 11f62 > $FFFF! Please submit a bug report!
_fprintf_snes_label(): Label's address inside a bank is 11fbc > $FFFF! Please submit a bug report!
_fprintf_snes_label(): Label's address inside a bank is 12036 > $FFFF! Please submit a bug report!
_fprintf_snes_label(): Label's address inside a bank is 12060 > $FFFF! Please submit a bug report!

```

The code in attachment reproduce it, please let me know if i need to provide you other files :)

vhelin, 2021-04-11:
Ok, try with the latest sources, you might get less of those TRY_PUT_LABEL's as the linker put all dead labels also to hash tables. I also improved the _fprintf_snes_label() error to print the label's name so you can find the problematic labels (inside libc.obj in your case)...

vhelin, 2021-04-11:
I'll look into _fprintf_snes_label() errors, but could you mean while check that there isn't anything wrong with the labels in libc.s?

vhelin, 2021-04-11:
I cannot find sm_spc in libc files.zip ... ?

vhelin, 2021-04-11:
libc.obj: sm_spc.asm:6: _fprintf_snes_label(): The address of label "SECTIONSTART_.sm_spc" inside a bank is $10490 > $FFFF! Please submit a bug report!
libc.obj: sprites.asm:37: _fprintf_snes_label(): The address of label "SECTIONSTART_.reg_oams7e" inside a bank is $11a41 > $FFFF! Please submit a bug report!
libc.obj: sprites.asm:45: _fprintf_snes_label(): The address of label "SECTIONSTART_.sprites0_text" inside a bank is $11c62 > $FFFF! Please submit a bug report!
libc.obj: sprites.asm:80: _fprintf_snes_label(): The address of label "SECTIONSTART_.sprites1_text" inside a bank is $11c92 > $FFFF! Please submit a bug report!
libc.obj: sprites.asm:124: _fprintf_snes_label(): The address of label "SECTIONSTART_.sprites2_text" inside a bank is $11cc6 > $FFFF! Please submit a bug report!
libc.obj: sprites.asm:273: _fprintf_snes_label(): The address of label "SECTIONSTART_.sprites3_text" inside a bank is $11f62 > $FFFF! Please submit a bug report!
libc.obj: sprites.asm:349: _fprintf_snes_label(): The address of label "SECTIONSTART_.sprites4_text" inside a bank is $11fbc > $FFFF! Please submit a bug report!
libc.obj: sprites.asm:440: _fprintf_snes_label(): The address of label "SECTIONSTART_.sprites5_text" inside a bank is $12036 > $FFFF! Please submit a bug report!
libc.obj: videos.asm:28: _fprintf_snes_label(): The address of label "SECTIONSTART_.pvsneslib_text" inside a bank is $12060 > $FFFF! Please submit a bug report!

vhelin, 2021-04-11:
If you reassemble everything with the latest WLA DX sources, do you still get those SECTIONSTART errors? I fixed a bug I just added in WLA regarding SECTIONSTART labels... :)

hoit, 2021-04-11:
i just built it and this error seems to be solved, no more printf_snes_label() error, thank you :)

I still have some issues but it is related to our code that is not finished (we tested only with ram.data section for now). I have to do some changes and more tests then i will keep you informed about it.

About sm_spc, do you still need i provide it to you ? This part build binary file with TASM tool (for SPC700), then export it to byte array directly in sm_spc.asm :
[sm_spc.asm.txt](https://github.com/vhelin/wla-dx/files/6292282/sm_spc.asm.txt)

I want to replace TASM by WLA-spc700 but i do not know if it is hard to do, i will take a look when the code will works with this version of wla !

vhelin, 2021-04-11:
Ah, I have no need for sm_spc any more.

> I want to replace TASM by WLA-spc700 but i do not know if it is hard to do, i will take a look when the code will works with this version of wla !

Let me know if you run into other issues.

hoit, 2021-04-16:
Hi,

By trying the last update (with the change on sections), a big project from Alekmaul produce this error with Wlalink then the application crash on win xp: 

OBTAIN_ROMBANKS: Using the biggest selected amount of ROM banks (64).

The crash of application seems to be related to this error, do you use return instead of exit in the code which could explain it ?

The (huge) project was not built before your last change and the error is probably not related to the last change. Could you explain what can be the cause of it ?

Last question: it is very hard to debug the wlalink part. How can we have more informations on all steps executed ? Is it possible to have a debug version which will provide more details and to see how are created .obj files or their content ?

Thanks!

vhelin, 2021-04-18:
> By trying the last update (with the change on sections), a big project from Alekmaul produce this error with Wlalink then the application crash on win xp:
>
> OBTAIN_ROMBANKS: Using the biggest [...]

This warning means that the object files in your project have different ROM sizes defined and WLALINK tries to use the biggest ROM size defined. Naturally this is not an ideal situation...

BTW, have you gotten smaller projects to work?

The function, that gives this warning, obtain_rombanks() in wlalink/analyze.c uses return... ?

> edit: on windows xp with command line it produce an error 99 and the c++ builder debugger write « access violation », maybe an array which is too short in wlalink? Because this project has a lot of sections with differents names
> 
> The (huge) project was not built before your last change and the error is probably not related to the last change. Could you explain what can be the cause of it ?

Please check that all object files in your huge project have the same ROM size defined. That should fix the warning and possibly the crash as well...

> Last question: it is very hard to debug the wlalink part. How can we have more informations on all steps executed ? Is it possible to have a debug version which will provide more details and to see how are created .obj files or their content ?

All the steps WLALINK does on main level are specified in wlalink/main.c, just check out the function names and comments. :) 

If you want a debug version of WLALINK:

  cmake -DGDB_DEBUGGING=ON -DCMAKE_BUILD_TYPE=Debug -G "[your platform]"

also make sure you have WLALINK_DEBUG defined at least in wlalink/main.c. Then WLALINK.EXE should print out quite a lot of information about the object files it reads in and labels and sections after it has placed them. If that's not enough then you just must and fprintf()s to WLALINK's code...

hoit, 2021-04-26:
> BTW, have you gotten smaller projects to work?

Yes, the samples provided with the pvsneslib library seems to work. I also found sources from a small game that works correctly but i think  these projects are not big enough to discover all issues.

I built wlalink with debug option, sorry i not saw before that this step is already mentioned in the install file...

The project which crash has been changed to remove the warning OBTAIN_ROMBANKS but the crash is still existing. It comes from this line in `int merge_sections(void)` in **analyze.c** file

```
/* sort the append sections by priority first and then by size, biggest first */
  qsort(ass, asn, sizeof(struct append_section *), _append_sections_sort);
```
  
I am looking how i can provide you more details because it is not my project and i do not have source code to test it easily

vhelin, 2021-04-26:
If the user doesn't append sections using priorities then you can comment out that line of code.

But it crashed? Any backtraces etc? :)

If that qsort() crashes I think it means that the sorting function _append_sections_sort() crashes and takes qsort() with it...

I'll try to make _append_sections_sort() more informative in case of problems, just give me an hour or two...

vhelin, 2021-04-26:
If you try now with the latest sources you should get more information about your problem... Hopefully. :)

hoit, 2021-04-27:
Great, thank you for this update. I confirm that wlalink does not crash anymore and i now have a lot of errors in the file output (like _find_append_source_section() and _appends_sections_sort()).
[log.zip](https://github.com/vhelin/wla-dx/files/6383664/log.zip)

The compilation finally failed with `maps2.obj: maps2.asm:865: PARSE_STACK: Unresolved reference to "objbuffers.objxpos"` but it seems to be declared and included, i guess it is related to previous errors

I do not know if it can be related to the order of files included in linkfile and i am checking how it can occurs. I keep you informed if i find anything :)

vhelin, 2021-04-27:
I think something is really broken as these errors

_find_append_source_section(): Section 6 from file "sound.obj" has gone missing. Please submit a bug report!
_find_append_source_section(): Section 7 from file "sound.obj" has gone missing. Please submit a bug report!
_find_append_source_section(): Section 7 from file "levels.obj" has gone missing. Please submit a bug report!
_find_append_source_section(): Section 8 from file "levels.obj" has gone missing. Please submit a bug report!
...

mean that those sections have gone missing along with the data and labels inside those sections... Why that has happened, I don't know, but it would be nice to know. 

If you try again with the latest sources you'll get better error messages at least. Linking also might work better, but I cannot guarantee it.

Please post the new log.zip here again!

vhelin, 2021-04-27:
Also a question about this big project that's failing: when was the last time it linked and worked, using which WLA DX? Does that same WLA DX still link it ok?

hoit, 2021-04-28:
Here is the new log.zip with the last version of wla : 
[log.zip](https://github.com/vhelin/wla-dx/files/6391050/log.zip)

> Also a question about this big project that's failing: when was the last time it linked and worked, using which WLA DX? Does that same WLA DX still link it ok?

This project has been built initially (and worked) with the old wla 9.4 version which was adapted (it was not the official one) to work with tcc816 (the c compiler) : i think it was this [version](https://github.com/optixx/snes-sdk/tree/master/wla_dx) 

Since we began the upgrade to the newer wla version, i think the rom has been generated (with sources before this [change](https://github.com/vhelin/wla-dx/commit/90749a5ca211c1340eedaaae0e8dfc02b977d9b3) ) but the game did not worked because of internal things to adapt in pvsneslib due to 9.4 upgrade (for instance the APPENDTO and KEEP instructions to have only one ram section). I will try to find the wla version where the project generate the rom.

On small projects and sample, we do not have these errors. This project has more source files, does the order in linkfile could explain that some objects are not found ?

vhelin, 2021-04-28:
Hmm, ok...

A quick question: Section "rams3egx.data" from file "sound.obj" - is there anything special about that section? Would it be possible to see the code/contents of that section?

All sections that _find_append_source_section() cannot find are .data sections... Are they empty?

vhelin, 2021-04-28:
> This project has more source files, does the order in linkfile could explain that some objects are not found ?

The order in linkfile should not matter as all sections are loaded in and then sorted by their priority first and size second. What bothers me is that some sections have gone missing after loading them. Or has WLA the assembler already dropped the section, but it stil writes the APPENDTO command to assembled object file? This could be the case.

If that's the case then perhaps I'll need to add a flag to WLA so that it doesn't drop empty sections? Or one can add KEEP to their definition...

vhelin, 2021-04-28:
I now made it so (in the latest sources) that if an empty section is using APPENDTO so that it'd be appended to another section by WLALINK and WLA drops the section, because it's empty, WLA now also drops the APPENDTO command as it's pretty useless.

I also added flag "-k" to WLA that forces WLA to keep all empty sections (and labels inside)... Perhaps you need that, or add KEEP to all empty sections you have?

hoit, 2021-04-28:
Yes you are right, the sections where the issue occur were empty because data have been moved from ram section to rodata because of const :

```
.RAMSECTION "rams4j0x.data" APPENDTO "globram.data"
.ENDS
.SECTION "s4j0x.data" APPENDTO "glob.data"
.ENDS
```

It is solved and these errors does not appear anymore with the last version :) but the final error before creating the rom is still produced:

`maps2.obj: maps2.asm:865: PARSE_STACK: Unresolved reference to "objbuffers.objxpos".`

Here are the last log with your last commit (log before and after APPENDTO)
[log.zip](https://github.com/vhelin/wla-dx/files/6394970/log.zip)

Does this log file can provide me some tracks on the issue ?

maps2.obj and objects2.obj are included in the linkfile.

For now, i do not find a way to reproduce it but if you have any idea or something i can check, please let me know :)

Thanks !

vhelin, 2021-04-29:
> maps2.obj: maps2.asm:865: PARSE_STACK: Unresolved reference to "objbuffers.objxpos".

Ok, this is easy: WLA doesn't generate in your case objbuffers.objxpos any more, unless there is just one instance of a structure. You'll need to change that reference to "objbuffers.1.objxpos".

hoit, 2021-05-02:
Hi !
I am very happy because everything now build correctly and seems to work too (without initialisation issue or anything).

We had some difficulties to understand us (probably because of my asm level which is bad or my explanations) but it finally work.

I want to say you a big thank you for your help and all improvements you did in wla for this issue, i can finally close it.

See you later!

closed 2021-05-02 by hoit

vhelin, 2021-05-02:
That's super great, and I'm really happy! This was quite a lot of work, but in the end your projects assemble and WLA DX just got a bit better - win-win!