vhelin/wla-dx #403: Compile decode_*.c into decode_*.o

vhelin, 2021-04-16:
Currently decode_*.c are included into pass_1.c. The code inside decode_*.c is just the contents of CPU instruction parsing loop.

Making decode_*.c to compile into decode_*.o (make decode_*.c contain only functions) needs to be done so we can more easily get rid of g_inz and g_ind references. And currently editing decode_*.c is a bit painful, because of indentation not working and the loop using variables from pass_1.c...

lhsazevedo, 2021-04-27:
List of globals used on `decode_*.c` files:

```sh
$ ls -1 decode_*.c | xargs grep -o '[^a-zA-Z0-9_]g_[a-zA-Z0-9_]*' | sed 's/\(.*\):.\(.*\)/\2/' | sort | uniq
g_accu_size
g_active_file_info_last
g_buffer
g_index_size
g_inz
g_label
g_latest_stack
g_operand_hint
g_operand_hint_type
g_opt_tmp
g_parsed_int
g_parse_floats
g_source_pointer
g_stacks_tmp
g_xbit_size

```

vhelin, 2021-04-27:
I guess I'm the best guy to do fix this issue as I've coded those decode_*.c files, and the main CPU instruction parsing loop... When I run out of more important things I'll fix this issue. :) Unless ofcourse someone else really wants to do this.

vhelin, 2022-10-02:
We now have these files inside decode.c so I think this issue is solved.

closed 2022-10-02 by vhelin