vhelin/wla-dx #412: Symbols for first and last used address in a SLOT

Kroc, 2021-05-08:
The BGB emulator has a useful feature that will break into the debugger if the code reads from a RAM address that hasn't been written to before; this is useful to catch errors with reading from unitialised memory. Most sample code out there, and even commercial games that have been disassembled, tend to wipe the whole of RAM during startup, but I'd like to only clear the RAM addresses actually used. Writing zeros to every known variable by name would waste a lot of code, so I'd prefer to be able to know the first and last address used in RAM without having to manually maintain the names of which variable is first/last.

Of course, slot usage is dependant on bank, so this would need to be factored into the symbol name. Perhaps `BANK0START_SlotName` or `SLOTSTART_SlotName_BANK0`.

vhelin, 2021-05-08:
Do you need the information as such definitions/labels, or would it be enough that WLALINK would print info about free space in RAM banks like it currently prints about free space in ROM?

Current output of WLALINK:

../../../binaries/wlalink -v -s linkfile linked.rom
Free space at $0028-$002a.
Free space at $006c-$01ff.
Free space at $023d-$3fff.
Bank 00 has 08026 bytes (97.97%) free.
Bank 01 has 08192 bytes (100.0%) free.
16218 unused bytes of total 16384.

Would it also be useful to change the current output to list free space per ROM bank? I'm thinking about this:

ROM bank 00
  - Free space at $0028-$002a
  - Free space at $006c-$01ff
  - Free space at $023d-$3fff
ROM bank 01
  - Free space at $0000-$3fff
RAM bank 00
  - Free space at $0010-$3fff

?

Kroc, 2021-05-09:
I need actual symbols, as I want to erase RAM from X to Y in the game's code, but I realise now that this requires very late information and might be difficult to implement

vhelin, 2021-05-09:
This is the order of things in WLALINK:

```
  /* insert sections */
  if (insert_sections() == FAILED)
    return 1;

  /* compute the labels' addresses */
  if (fix_label_addresses() == FAILED)
    return 1;

  /* INSERT HERE RAM BANK USAGE LABEL CREATION */

  /* generate _sizeof_[label] definitions */
  if (g_create_sizeof_definitions == YES) {
    if (generate_sizeof_label_definitions() == FAILED)
      return 1;
  }

  /* sort anonymous labels to speed up searching for them */
  if (sort_anonymous_labels() == FAILED)
    return 1;

  /* compute pending calculations */
  if (compute_pending_calculations() == FAILED)
    return 1;

  /* transform computation stack definitions to ordinary definitions */
  if (transform_stack_definitions() == FAILED)
    return 1;

  /* fix references */
  if (fix_references() == FAILED)
    return 1;
```

I've marked the suitable spot for RAM bank usage labels creation.

It should be pretty easy to implement if it only uses RAM section information. Using .ENUM information is impossible, at the moment, as .ENUM generates just normal .DEFINEs and they are mixed with other .DEFINEs... Do you use only .RAMSECTIONs for RAM variables?

Kroc, 2021-05-09:
Yes, because of the hard RAM/ROM split in the GameBoy / SMS memory map.

vhelin, 2021-05-09:
Ok, now if you use the latest sources WLALINK will create labels RAM_USAGE_SLOT_x_BANK_y_START and RAM_USAGE_SLOT_x_BANK_y_END. Instead of creating symbols I made them labels so you can actually reference them from assembly code. Do they fix this issue?

closed 2021-05-25 by vhelin