vhelin/wla-dx #413: Right curly brace ("}") no longer allowed in strings?

SolraBizna, 2021-05-15:
The following trivial source file:

```asm
.DB "}"
```

Now gives the following error, with all of the architectures I tried:

```
test.65c:1: INPUT_NUMBER: String wasn't terminated properly.
test.65c:1: ERROR: Couldn't parse ".DB".
```

This error appears to occur when a `}` occurs anywhere in a `.DB` string.

vhelin, 2021-05-15:
I found some weird curly braces handling in parse.c, and couldn't figure out why it was there so I removed it, which should fix this issue. Perhaps it was part of {\@+1} handling in .MACROs one time, but this was inside a string ("") and didn't do the things .MACRO parsing would require, like a half finished implementation.

I hope I didn't break anything. :D

SolraBizna, 2021-05-15:
Confirmed fixed. Thanks!

closed 2021-05-15 by SolraBizna