vhelin/wla-dx #414: @-local labels can no longer be passed to macros by value

SolraBizna, 2021-05-15:
If I have understood correctly, commit 719e95e adds a feature whereby preceding a macro parameter with `@` passes the given name, not its value.

Here is a sample of no-longer-working code, distilled out of an existing codebase of mine:

```asm
.MEMORYMAP
DEFAULTSLOT 0
SLOT 0 $0000 $10000
.ENDME
.ROMBANKSIZE $10000
.ROMBANKS 1

.MACRO PtrSet ARGS ptr, val
	LDA #<val
	STA ptr
	LDA #>val
	STA ptr+1
.ENDM

_start:
	PtrSet my_ptr, @my_val
@my_val:
.DB "Hello World!"

my_ptr:
.DW $0000
```

(I feel like if I understood why GitHub colored that with an Easter theme, I would be much less confused about life in general.)

Error from compiling and linking that example:

```
test.o: test.65c:9: PARSE_STACK: Unresolved reference to "my_val".
```

Note that the linker is looking for `my_val` without the `@`. If I modify the code to pass `@@my_val` to `PtrSet` instead, it compiles to the proper result. (In this example—and all the usages in my real codebase—nothing has been lost by passing the name instead of the value.) When I first encountered this issue, I tried escaping the `@` as `\@` (which just made the `\` part of the label) and passing `0+@my_val` (which failed because `@my_val` isn't known yet) and `@my_val+0` (which seemed to be passing the name `my_val` again).

Perhaps a different notation should be used for this feature? Or maybe some diagnostic to catch existing code that passes @-locals to macros and is now broken, so as to suggest the `@@` workaround?

vhelin, 2021-05-15:
Ouch, I forgot those child labels (that begin with @) completely. @@ would not work as 2nd level child labels begin with @@. I'm thinking about changing @ here to &...

SolraBizna, 2021-05-15:
Confirmed fixed. My original code now builds again.

I can't think of any problems with using `&` instead, but one of the downsides of having a powerful, feature-rich assembler is that it's easy to forget about one of the features.

(By the way, thanks for making such a powerful, feature-rich assembler.)

closed 2021-05-15 by SolraBizna

vhelin, 2021-05-15:
Good to hear original code builds again. And thank you, it's been a lot of work, but also challenging and partially fun. :)