vhelin/wla-dx #415: With -A, wlalink produces [source files] with 16-digit checksums half the time on LP64

SolraBizna, 2021-05-23:
When outputting the checksum for files under the `[source files]` section, `%.8lx` is used as the format specifier. All well and good. Except that, on an LP64 platform (where unsigned long is 64 bits), sometimes the checksum being outputted has been sign-extended, which leads to 16 digits actually being written if bit 31 was set. Example:

```
0001 3667fb73 src/console.65c
0002 fffffffffbb1d20a include/common.inc
```

I'd suggest bitwise ANDing with `0xFFFFFFFF` (or `0xFFFFFFFFUL` if supported on all targets?) as an easy fix.

This isn't very high priority, as it's trivial to ignore the extra bits on parsing. Also, the checksum under `[rom checksum]` does not appear to be affected by the same problem.

vhelin, 2021-05-23:
Good find! crc32() does internally already &0xffffffffUL so that's why [rom checksum] works.

SolraBizna, 2021-05-23:
Confirmed fixed. Thanks!

closed 2021-05-23 by SolraBizna