vhelin/wla-dx #416: With -A, wlalink not producing unique IDs for different source files

SolraBizna, 2021-05-23:
It appears that the original source file, and all included files, are assigned a unique ID by the assembler, which is then passed through without further processing by the linker. Which results in situations like the following:

```ini
[source files]
0001 bd2efbd6 src/bcd.65c
0002 fbb1d20a include/common.inc
0003 2fb666c3 include/pub/ars_registers.inc
0004 ae5f9759 include/pub/stdint_globals.inc
0005 04a5287d include/pub/dos_vars.inc
0001 8c000200 src/boot.65c
0002 fbb1d20a include/common.inc
0003 2fb666c3 include/pub/ars_registers.inc
0004 ae5f9759 include/pub/stdint_globals.inc
0005 04a5287d include/pub/dos_vars.inc
0001 fdaa1b28 src/builtins.65c
0002 fbb1d20a include/common.inc
0003 2fb666c3 include/pub/ars_registers.inc
0004 ae5f9759 include/pub/stdint_globals.inc
0005 04a5287d include/pub/dos_vars.inc
...
```

(the effects of #415 have been manually trimmed from the above for clarity)

Later, in the `[addr-to-line mapping]` section, these non-unique unique IDs are used, making it impossible to tell which symbols come from which files.

It seems that the linker needs to separate out these IDs before writing them, possibly coalescing duplicates in the process. (It's possible for the same included file to have a different meaning when included in a different source file, but I'm of the opinion that if you use that technique, you should *expect* to confuse your tooling. :) ) An advantage of *not* coalescing duplicates in this case is that it would be trivial to expand this format to incorporate "included-from" information, so that a given file ID actually denotes a particular "file X included from file Y included from ...", which would make debugging certain kinds of macro easier.

vhelin, 2021-05-23:
The source file IDs here are actually object files' internal IDs for source files, no idea why it's been done that way as they are not globally unique.

Would it be ok if I just added the object file ID to [source files] and [addr-to-line mapping]?

Examples:

[source files]
0001:0001 bd2efbd6 src/bcd.65c
0001:0002 fbb1d20a include/common.inc
0001:0003 2fb666c3 include/pub/ars_registers.inc
0001:0004 ae5f9759 include/pub/stdint_globals.inc
0001:0005 04a5287d include/pub/dos_vars.inc
0002:0001 8c000200 src/boot.65c
0002:0002 fbb1d20a include/common.inc
0002:0003 2fb666c3 include/pub/ars_registers.inc
0002:0004 ae5f9759 include/pub/stdint_globals.inc
0002:0005 04a5287d include/pub/dos_vars.inc

Here the first number is the object file ID, the second number is the source file ID inside that object file...

[addr-to-line mapping]
00:0000 0001:0001:00000015
00:0002 0001:0002:00000003
00:0004 0001:0003:00000003
00:0006 0002:0001:00000015
00:0008 0002:0002:00000003
00:000a 0002:0003:00000003

Here the 3rd number is the object file ID, the 4th number is the source file ID inside that object file...

vhelin, 2021-05-23:
Currently we don't have information about which source file was included from which source file, only which source files are inside which object files. If you need the source-included-from-source information then we'll need to add more bits into the object files...

SolraBizna, 2021-05-23:
That solution would work fine for me. In practice, just knowing which top-level source file is involved would be enough for any application I've written (SO FAR :D).

vhelin, 2021-05-23:
Ok, it should be done now. :)

SolraBizna, 2021-05-24:
Looks great. But now I'm out of excuses to avoid implementing the next chunk of address mapping plumbing in my emulator. :(

closed 2021-05-24 by SolraBizna