vhelin/wla-dx #421: Byte Tester: Allow multiple source files

lhsazevedo, 2021-07-04:
[Some tests](https://github.com/vhelin/wla-dx/tree/master/tests/6502/multifile_test) check the output of multiple source files, and cannot be tested using the new source file option.

They could be if we allow multiple source files on byte tester.
The tests instructions of each file would simply be concatenated as if it was one big file, example:
```
main.s:

@BT result.rom
.db "01>"     // @BT TEST-01 01 START
ld a, (hl)    // @BT DE AD

-------

more.s:

ld c, a       // @BT BE EF
.db "<01"     // @BT END
```

```$ byte_tester -s main.s more.s```

Tests instructions read by byte tester:
```result.rom TEST-01 01 START DE AD BE EF END```

vhelin, 2021-07-05:
Sounds good!

vhelin, 2023-02-22:
I don't think that would work on an Amiga where I use the run_tests.sh script to run the tests. There the shell is not a modern Bash shell, and it's pretty buggy and old. That said I think we should make this by enhancing byte_tester as it shouldn't be a huge task and would then work everywhere regardless of the shell.

closed 2026-03-05 by vhelin