vhelin/wla-dx #422: Convert tests to the new in-source tests instructions feature

lhsazevedo, 2021-07-04:
Keeping the test instructions inside the source file greatly reduce the amount of context switching when authoring and reviewing tests.

Recently, the Byte Tester gained the ability to read the tests instructions from the sources itself using the `-s` flag. The [z80 mnemonics_test](https://github.com/vhelin/wla-dx/blob/master/tests/z80/mnemonics_test/main.s) already uses this feature. 

If you are OK with this, I can convert the remaining tests to this new mode.
Note that tests that checks the output of multiple source files will be converted after is implemented #421.

vhelin, 2021-07-05:
I'm ok with this! :)

vhelin, 2023-11-12:
My commit here https://github.com/vhelin/wla-dx/commit/08a494f1aaea05d37aab04b3897092b5613b914d converted the last tests so we can close this issue.

closed 2023-11-12 by vhelin

lhsazevedo, 2023-11-12:
Nice! I hope this made writing tests less boring

vhelin, 2023-11-12:
It's convenient to have the test data right next to the code it tests, but my main motivation for converting the last tests was that on Amiga my test runner script cannot use testfiles, but the embedded tests work just fine. Now that I was having trouble with the Amiga build I wanted to run all the tests on Amiga that I could...