vhelin/wla-dx #429: Option to restrict .SECTION placement to not cross pages

ToriningenGames, 2021-08-07:
This is a proposed enhancement.

Sometimes, I have a section containing mainly data, whether via `.INCLUDE` or `.DB`. The section's size is less than 256 bytes.
Therefore, if I fit the section within a page (to use 6502 terminology; the concept applies to most 8 bit processors), I can access the data more quicky, since the upper byte of the address of all the data is constant; for loops, I only need to increment/decrement the lower byte of the address.
This is doable using `ALIGN 256`, but it is unnecessarily strict; sometimes it takes no extra time or space if the data isn't page aligned, so more compact output could be generated by slipping it right after another section, before the page ends.

Therefore, I propose a `WINDOW` argument to sections, that would ensure the section is placed fully with such an area. It would take an integer as an argument, allowing for arbitrarily sized windows. So, `WINDOW 256` would keep all data with the same page, while `WINDOW 4096` would place all data in a 12 bit window (all addresses would come out as, say, `$C???` or `$3???`) This would allow other processors with other sized ranging limits to be accomodated (the Motorola 68000 comes to mind, since keeping everything within a 64k window makes a lot of stringlike operations more straightforward)

An alternative option would be to specify how many bits would be allowed to vary, so the above two examples turn into `WINDOW 8` and `WINDOW 12`. While clearer, this would eliminate nonbinary windows, and would be inconsistent with `ALIGN`.

vhelin, 2021-08-07:
Sounds like a good feature. "WINDOW [START] [END]" perhaps? Could also add something like "BITWINDOW [bits]" while implementing this feature...

ToriningenGames, 2021-08-07:
Compatibility with `FREE` and `SUPERFREE` would be nice, so that libraries could also use it. The boundries are the important parts.

ToriningenGames, 2021-08-10:
I appreciate the nontrivial placement modification. Automatic address assignment are one of the things that make Assemblers great.

At this point, I see conceptual similarity between this and banks & slots. The variables are different, so I don't know how much code can be reused, but it's like section specific subslotting.

vhelin, 2021-08-15:
Could start working on this today a little past 22:00, had quite a busy week end, so I guess I'll have to work on this also next week... :P

vhelin, 2021-08-16:
You can use WINDOW and BITWINDOW with libraries, but only via WLALINK's linkfile. As .SECTIONs/.RAMSECTIONs in a library don't take SLOT/BANK when building, I thought that WINDOW/BITWINDOW in a library file would be unwise as library .SECTIONs/.RAMSECTIONs need their entries in linkfile anyway... If you think this is not good, let's discuss it here. :)

And I hope this works, I'm a bit tired. I wrote a couple of tests for WINDOW and BITWINDOW, but there might be something I've missed...

ToriningenGames, 2021-08-16:
An argument could be made against it, since BITWINDOW operates like ALIGN (which can be put on free sections in libraries), and its use is necessitated by the code, so having it in the linkfile separates it from the code, so it's harder to get a gague on the code's restraints and privileges.

However, if it works, it works, and the above concerns can be addressed by process. I'm honestly not familiar with WLA's linking process, so I can't say what's easy or doable (I just remember that in 9.0, a cursory examination told me 32 and 64 bit linking would be Very Hard)

vhelin, 2021-08-16:
Good point. Let me drink a beer and think about this, I'll be back in 20 minutes... :)

vhelin, 2021-08-16:
Yeah, well, if the user wants to use WINDOW/BITWINDOW in a library file, why block that... :)

PS. Ouch. I forgot to change the library file format... Gotta fix that, too...

vhelin, 2021-08-16:
Let us know if you have any problems with WINDOW/BITWINDOW, I'm not closing this issue right away...

vhelin, 2021-08-22:
@ToriningenGames Did you test this new feature, does it work? :)

ToriningenGames, 2021-08-23:
Sorry I've been away on business and have limited access to my setup. I should be able to try it tonight.

ToriningenGames, 2021-08-24:
.BITWINDOW works in my case (am on Windows) and is what I had hoped for. Doesn't help save space, but consolidates free space into one blob.
Couldn't use .WINDOW... because I don't know its syntax and didn't see it in the documentation (and I'm too lazy to read the source lol). Making the docs has always been wonky for me though; cmake doesn't play nice with nonstandard install locations not on PATH, and Python version incompatibilities are... well...
Basically I probably missed it.

vhelin, 2021-08-24:
Just search for WINDOW in doc/asmdiv.rst - that's plain text, very readable, and source for the .html documentation...

ToriningenGames, 2021-08-25:
Either I'm insane, or I'm insane and it's not there.

vhelin, 2021-09-10:
Ok, I assume this issue is now fixed.

closed 2021-09-10 by vhelin