vhelin/wla-dx #435: Runinng tests under Cygwin (latest binaries 22-Sep-2021) prints weird texts

vhelin, 2021-09-22:
I just updated my Cygwin installation, and now when I run the tests I get two weird lines:

[image]

From the dots you should be able to reason which tests create problems...

vhelin, 2021-09-22:
Wait a sec, that run had debug executables. With release executables I don't get that weird behavior:

[image]

I'll check again if make clean resolved this issue with release executables...

vhelin, 2021-09-22:
Yes, debug executables still give me those weird two lines... Attached is the output of "./run_tests.sh -v &> output.txt"...

[output.txt](https://github.com/vhelin/wla-dx/files/7213457/output.txt)

Happy hunting. XD I think it might be better just to count which tests generate the errors and try them individually, but I'm running out of time today...

vhelin, 2021-09-24:
I was able to shrink the run_tests.sh into this, which gives with debug versions of WLA executables the warning:

```

#!/bin/sh

#
# NOTE! All the small and ugly projects inside the "tests" directory
# are here used as unit tests / if the projects assemble and link without
# errors, we'll assume that WLA DX is in ok shape. Feel free to add more
# test projects to the lot.
#

set -e

runTest() {
    set -e
    cd $1
    make clean
    make
    if test -f "testsfile"; then
        eval "$BYTE_TESTER testsfile"
    fi
    make clean
    cd ..
}

cd tests
cd 6502

OUT=$(runTest functions_test 2>&1) # THIS GIVES THE WARNING

```

My Bash sucks, so I don't know where that null byte is. Command substitution?

lhsazevedo, 2021-09-24:
The $(...) thing is called command substitution. We use it to capture the current test output and store it in a variable (OUT), so we can display it in case of failure.

I think that make, byte_tester or wla itself are printing a null byte, and bash is just telling us that this null byte is being ignored.

We can fix this by tracking down the null byte and preventing it from getting printed, or just strip it from the output as [this answer](https://askubuntu.com/a/926695) suggests.

vhelin, 2021-09-24:
AFAIK null bytes don't really belong to shell output so best would be to find the source where the null byte is printed and fix that. But now we know the test that generates the error...

vhelin, 2021-09-24:
Great tip!

[image]

It seems the problems might be in stack.c/_debug_print_stack() ?

lhsazevedo, 2021-09-24:
Looks like it is! My bet is line 141:
https://github.com/vhelin/wla-dx/blob/371b2edbb9dedbadca54df07bae032c1b67232f9/stack.c#L136-L142

If confirmed, should we print it as hex?

vhelin, 2021-09-24:
Hmm could it be that that function is missing some new SI_OP_*s? I'll check that out later today, just busy right now. :)

vhelin, 2021-09-24:
Yep. It's missing three new SI_OP_*s (the last three ones in shared.h):

[image]
[image]

... and it also seems that WLALINK is not handling these or some even older SI_OP_*s in its debug output...

Anyway, I'll fix this right now with the help from a Friday beer. :)

vhelin, 2021-09-24:
Good thing we investigated further and not just filtered out the null bytes. :) We found some real issues!

vhelin, 2021-09-24:
Should work now :)

closed 2021-09-24 by vhelin