vhelin/wla-dx #440: Possibly incorrect expectation in Super FX "Move (xx), Rn" test

lhsazevedo, 2021-09-27:
According to SNES Development Manual, the [MOVE (xx), Rn](https://archive.org/details/SNESDevManual/book2/page/n241/mode/1up) macro instructions at TEST-12 must be replaced with [SM (xx), Rn](https://archive.org/details/SNESDevManual/book2/page/n268/mode/1up) and [SMS (yy), Rn](https://archive.org/details/SNESDevManual/book2/page/n269/mode/1up) instructions as follows:

```
move ($A1), r10             ; sm ($A1), r10
move ($A2), r11             ; sms ($A2), r11
move (label_a), r12         ; sms (label_a), r12
move (label_a | $1200), r13 ; sm (label_a | $1200), r13
```

And assembled as:

```
sm ($A1), r10              ; 3E FA A1 00
sms ($A2), r11             ; 3E AB 51
sms (label_a), r12         ; 3E AC 11
sm (label_a | $1200), r13  ; 3E FD 22 12
```

But currently our tests are:
```
move ($A1),r10              ; 3D FA A1 00
move ($A2), r11             ; 3E FB 51
move (label_a),r12          ; 3D FC 22 00
move (label_a | $1200), r13 ; 3D FD 22 12
```

Did I miss something here? Or we are indeed expecting the wrong output?

Related issue: https://github.com/vhelin/wla-dx/issues/390

vhelin, 2021-09-27:
It could be that I've made a mistake there. I've got zero feedback on the Super FX support in WLA DX...

lhsazevedo, 2021-09-27:
If there is a mistake, laying the tests side by side with the code definitely helped me spot it

lhsazevedo, 2021-10-10:
@Ramsis-SNES Would you mind to help confirming this please?

lhsazevedo, 2021-10-11:
Then I'll update https://github.com/vhelin/wla-dx/pull/439 to fix the tests as well

vhelin, 2022-01-17:
This has been hanging here for so long that I need to take care of it asap, like this week. Sorry for the delays, I've been really busy elsewhere, but now have time for WLA DX again as well...

lhsazevedo, 2022-01-17:
No worries, for me at least. I just found this when converting some tests, but I don't use the Super FX assembler.

vhelin, 2022-01-17:
I've gotten 0 reports from people _using_ the SuperFX assembler so far... :D

vhelin, 2022-02-04:
Finally I had time to fix this. And by fix I mean it should work now. Not as well as defined in SNES Development Manual Book II, but it should work...

I hope you @lhsazevedo can continue the good work you've been doing so far. Without your sharp eyes IDK when this would have been spotted. :)

vhelin, 2022-02-04:
For reference, here is TEST-12:

```
 .db "12>"
        .print "12>\n"
        move ($A1),r10              ; -> sm ($A1), r10   -> 3E FA A1 00
        move ($A2), r11             ; -> sms ($A2), r11  -> 3E AB 51
        move (label_a),r12          ; -> sm ($22), r12   -> 3E FC 22 00 ... NOTE: as WLA doesn't know that label_a is $22 it bails out to sm here...
                                    ;                                       this is not what SNES Development Manual says, but then again WLA DX
                                    ;                                       is not an official Nintendo/SNES assmebler...
        move (label_a | $1200), r13 ; -> sm ($1222), r13 -> 3E FD 22 12
        .print "<12\n"
        .db "<12"
```

vhelin, 2022-02-04:
The problem with the 3rd row is that WLA doesn't know that label_a will become $22 after WLALINK has placed the label into its final resting place. So WLA cannot turn MOVE macro instruction here into SMS...

I know this is not what SNES Development Manual Book II says, but WLA is not that smart and WLA is not the official Nintendo SNES assembler, so I guess this is the best we can do at the moment.

I could force WLA to output 4 NOPs and make WLALINK to choose between SM and SMS, but in the case of SMS we'd still have one NOP in the output - not an ideal solution either.

cr1901, 2022-02-04:
I wonder if ARGASM/ARGLINK had linker relaxation to support turning MOVE into SMS...

closed 2022-02-10 by vhelin