vhelin/wla-dx #446: Makefile generation issues

maxim-zhao, 2021-12-15:
I have been using makefile generation to improve my build process but I have found a few issues... here's an example input:

```
.memorymap
defaultslot 0
slot 0 $0000 $4000 ; ROM
.endme

.rombankmap
bankstotal 1
banksize $4000
banks 1
.endro

.block "foo"
.db 0
.endb

.include "foo\bar.asm"
.include "foo\t1.asm"
.include "foo/bat.asm"
.include "foo/t2.asm"

.incbin "foo\bar.bin"
.incbin "foo\t1.bin"
.incbin "foo/bat.bin"
.incbin "foo/t2.bin"
```
Resulting in:
```
INTERNAL_PASS_1: Block "foo" is 1 bytes in size.
foo.o: test.asm \
	foo\bar.asm \
	foo	1.asm \
	foo/bat.asm \
	foo/t2.asm \
	foo\bar.bin \
	foo	1.bin \
	foo\bat.bin \
	foo\t2.bin
```

1. Block sizes go to stdout
Thus the makefile is not valid.

2. "\t" in paths is converted to a tab
Presumably other control characters get parsed too. So maybe I can use forward slashes throughout as a workaround...

3. "/" is converted to "\" for `.incbin` but not `.include`, if I run on Windows
This inconsistency gets confusing, Now I need to make sure my makefile patterns work for both types of slash.

Not in the example...

4. Some syntax using external files doesn't play properly with this mode

Notably `.stringmaptable`, which I added. Example syntax:

`.stringmaptable foo bar.tbl`

If bar.tbl does not exist, the build fails. If it does, it is not emitted to the makefile list. Can you offer some guidance on how to support this?

`.fopen` and `.background` have the same issue. `.include` and `.incbin` work as expected.

vhelin, 2021-12-15:
Too busy at the moment to work on WLA DX, will have more free time 24-31st this month, will return to this then...

maxim-zhao, 2021-12-15:
My immediate fix was to use forward slashes throughout (asm and makefile), and have a post-processing step on the makefile generation to filter and correct the slashes; Python code:
```python
def fix_makefile(path):
   with open(path, 'r') as f:
       lines = f.readlines()
   with open(path, 'w') as f:
       f.writelines([re.sub('\\\\(.)', '/\\1', x) for x in lines if not 'INTERNAL' in x])
```

closed 2021-12-18 by maxim-zhao