vhelin/wla-dx #447: Unions cause line numbers to be misreported

maxim-zhao, 2021-12-15:
Source:
```
.memorymap
defaultslot 0
slot 0 $0000 $4000 ; ROM
.endme

.rombankmap
bankstotal 1
banksize $4000
banks 1
.endro

.enum $0

  .union
  foo db
  .nextu
  bar db
  .endu

.ende

.fail
```

```
$ wla-z80 -M -o foo.o test.asm
test.asm:20: HALT: .FAIL found.
```

`.fail` is on line 22, not line 20. It is correctly reported if I place it before the union.

vhelin, 2021-12-15:
I guess that whoever added support for .UNION didn't advance the line number counter when the parser encounters a newline byte while parsing .UNION. I bet there are quite a few places in the .UNION parser where a newline byte can happen...

maxim-zhao, 2021-12-16:
Oddly there's three .union directives but the line count is off by two; I couldn't figure out how this could happen as none of them seems to advance the counter (I guess by calling `next_line()`?) so I didn't guess at a fix.

vhelin, 2021-12-18:
Yeah, next_line() is the function to call when parsing INPUT_NUMBER_EOL (end of line):

```
    q = input_number();
    if (q == INPUT_NUMBER_EOL)
      next_line();
    else ...
```

The code for .UNION in this example has two lines that are the "meat", not directives, perhaps the 2 offset in the line count comes from there - both of those lines end to an INPUT_NUMBER_EOL... ?

```
  .union
  foo db
  .nextu
  bar db
  .endu

vhelin, 2022-01-17:
@maxim-zhao Have you had time to test the fix? I'll close this issue soon if I don't hear that the fix doesn't work... :)

maxim-zhao, 2022-01-19:
It looks good to me.

Before:
```
$ make
wla-z80.exe -M -D LANGUAGE=en -D WLA_DX_BUG_WORKAROUND -o ps1jert.sms.o ps1jert.sms.asm > generated/auto.makefile
ps1jert.sms.asm:271: HALT: .FAIL found.
make: *** [makefile:103: generated/auto.makefile] Error 1
```
After:
```
$ make
wla-z80.exe -D LANGUAGE=en -o "ps1jert.sms.o" "ps1jert.sms.asm"
ps1jert.sms.asm:273: HALT: .FAIL found.
make: *** [makefile:109: ps1jert.sms.o] Error 1
```
```

closed 2022-01-19 by maxim-zhao