vhelin/wla-dx #453: Attempt to .DSTRUCT two of the same .STRUCTs results in a WLA-Z80 error `Label "<struct field>" was defined for the second time.`

SavagePencil, 2021-12-22:
The following code:
```
.STRUCT sState
    OnEnter DW
.ENDST

.DSTRUCT State1 INSTANCEOF sState VALUES:
    OnEnter:  .DW $1234
.ENDST

.DSTRUCT State2 INSTANCEOF sState VALUES:
    OnEnter:  .DW $5678
.ENDST
```

...generates the following error when compiled with `wla-z80` in version 10.1a:  ```INTERNAL_PASS_1: Label "OnEnter" was defined for the second time.```

SavagePencil, 2021-12-22:
This issue does NOT occur in 10.0.

vhelin, 2021-12-22:
I'll give this a look at the end of this week - currently I'm too busy with other things...

vhelin, 2021-12-24:
It seems that your code is broken, you have "VALUES:" there instead of "VALUES". That kinda breaks the assembler which then does silly things and gives a wrong error message.

I'll update the assembler to work better in this situation...

SavagePencil, 2021-12-24:
Aha!  Note that it does assemble correctly with 10.0.

vhelin, 2021-12-24:
v10.0 was more tolerant with incorrectly written directives? :) It's interesting why the parser parsed incorrect code and didn't fail...

vhelin, 2021-12-25:
Ok, v10.1 should now spot such an error in syntax and give a better error message.

closed 2021-12-25 by vhelin