vhelin/wla-dx #459: Substitute value of .DEFINE'd variable into a label name

Drenn1, 2022-01-06:
Hi, it's been a while. Half-question, half-request here. I'm wondering if it's possible to substitute a define to be used as part of a variable name.

Basically, I wanted to do something like this - using the rept "COUNT" variable in a label name:
```
.define NUM_TABLE_ENTRIES 2

myTable:
.rept NUM_TABLE_ENTRIES INDEX COUNT
	.dw tableEntry_{COUNT}
.endr

tableEntry_0:
    [...]

tableEntry_1:
    [...]
```
It seems this isn't possible, although it _is_ possible to use macro parameters and ``\@`` in this way:
```
.macro HELPER_MACRO
	.dw tableEntry_\@
.endm

.rept NUM_TABLE_ENTRIES
	HELPER_MACRO
.endr
```

The above code does work, but it's almost too silly for me to use =)

So, clearly it's _possible_ for a label name to have its value calculated dynamically like that, but it only works in limited cases such as the macro ``\@`` operator, and doesn't seem generalizable.

Full disclosure, my main project is currently on WLA v9.11 as I haven't gotten around to updating my project for v10+. I checked the documentation for newer versions and didn't see something resembling such a feature. Although I did see this intriguing example code in the documentation for ".MACRO":

```
.MACRO .test ARGS str
    __\._{\@+1}: ; this will become __.test_1 during the first call, __.test_2 during the second call...
```

This syntax isn't really documented beyond this one example, and as far as I can tell it seems limited to this specific use-case and not really generalizable. For example "1+1" does not get resolved to "2". Perhaps this could be expanded upon for not just arithmetic, but also string substitution - so whatever gets put in the "{}" brackets gets substituted as part of the variable name. That would potentially break any code that uses "{" in label or define names, although that's a bit of a heathenous thing to do anyway.

Drenn1, 2022-01-06:
Huh, so this seems about identical to discussion #457. I didn't notice at first since I normally just look at the issues list. Interesting.

vhelin, 2022-01-06:
I'm currently in crunch mode, we have a project deadline at work on 11-Jan-2022 evening, so I cannot look into this issue before that. But yeah, that __\\._{\\@+1} is just a special case and doesn't do much else than just that. It could be made so that if the contents in {} begin with something else than \\@ then it could work in another way, in a more general way...

vhelin, 2022-02-05:
Not sure if adding substitution for strings (in addition to labels) as well was wise, but please do comment. :)

Drenn1, 2022-02-05:
Some thoughts...

- Is it still possible to use curly brackets in strings using ``\{``? As I mentioned in a comment already I think there should be a test case for it.
- This doesn't appear to support substitutions used in ".EXPORT" directives, and perhaps others.
- You may want to remove the old special-case code for ``{\@+1}`` (and perhaps update the documentation for that)
- At the risk of feature creep... it would be nice if there were a way to specify whether to substitute a decimal or hexadecimal value. Most of my numbering is done in hexadecimal. Perhaps, ideally, with format strings (this would allow me to specify that they should always have two digits, ie. "02" instead of "2":
.dw {"label_index_%.2x", INDEX}
- Alternatively, since substitutions appear to work with definitions with "string" values, if there were some syntax to create strings with flexible format strings like that, then that could be an alternative. Example:
.definef my_string "label_index_%.2x", INDEX
.dw {my_string}

...Am I being too greedy? Let me know.

vhelin, 2022-02-05:
You can still use curly brackets in strings. Like "HELLO{COUNTER}" -> "HELLO{COUNTER}" if there is no definition called COUNTER.

Ouch, gotta check out what parsing functions .EXPORT etc. call, will need to add substitution to those as well...

No need to remove old special case \{\\@+1\} as it should still work.

Those nice-to-have-string-formatting-features - perhaps later. :)

Drenn1, 2022-02-05:
I am strongly against ``HELLO{COUNTER}`` working if there is no definition called ``COUNTER``. It should produce an error in this case.

I understand that this can break compatibility. But the alternative is having the ambiguity of not knowing what the purpose of the curly brackets is.

Is there any reason at all to keep code like this: https://github.com/vhelin/wla-dx/blob/master/parse.c#L1455? It can only be a maintenance burden if the new code does the same thing in a more general case. The reason I suggested using curly brackets in the first place was that I thought this was an unelegant special-case which could be replaced with more general-purpose code.

vhelin, 2022-02-05:
How should we then make old code work that has strings like "HELLO{COUNTER}" as now the new substitution code would fail for those strings? I don't want to make "\\{xyz\\}" to be the way of non-substitution. Perhaps we should do the substitution when string like "HELLO{SUBS(COUNTER)}"?

The reason to keep code like this: https://github.com/vhelin/wla-dx/blob/master/parse.c#L1455 - I don't want to remove a working feature that people might be using. As I'm doing 95%+ of the coding here I feel like I can keep on maintaining that special case... :)

vhelin, 2022-02-05:
"HELLO{%sCOUNTER}" ? That could be later expanded to "HELLO{%.2xCOUNTER}"...

Drenn1, 2022-02-05:
Oh, I see. This feature doesn't do arithmetic so it's not a replacement for ``{\@+1}``.

Perhaps you could consider allowing this feature to do arithmetic in order to replace the old one. Even if it doesn't affect me personally, the way that feature is implemented just rubs me the wrong way. It's arithmetic that doesn't use the arithmetic parser. I was hoping it could be obsoleted - not necessarily removed, just replaced with something more general-purpose that doesn't change the syntax.

As for the code failing on "HELLO{COUNTER}" - perhaps you can just disable this feature when used within quotation marks. I don't see any real need for it. So: ``.db HELLO{COUNTER}`` will always just produce the string ``"HELLO{COUNTER}"`` regardless of the "COUNTER" variable, and this feature will only apply when used on labels.

Drenn1, 2022-02-05:
To clarify: The original idea I had for this feature was that whatever was inside the curly brackets would be evaluated, and then substituted into the label. So I don't think the idea of allowing arithmetic inside the contents of the curly brackets should be considered too far-fetched.

vhelin, 2022-02-05:
> Perhaps you could consider allowing this feature to do arithmetic in order to replace the old one. Even if it doesn't affect me personally, the way that feature is implemented just rubs me the wrong [...]

Not a high priority issue I think.

> As for the code failing on "HELLO{COUNTER}" - perhaps you can just disable this feature when used within quotation marks. I don't see any real need for it. So: .db "HELLO{COUNTER}" will always just [...]

Ok, removed support for strings inside quotation marks

Drenn1, 2022-02-05:
Alright. Perhaps replacing ``{\@+1}`` can be considered for the future. Currently I guess this feature implemented as a simple hashmap lookup, but that could be expanded later to use the stack parser if there's ever desire for that... I guess. Not like I know how difficult it may be.

Anyway, as it stands right now I don't really have any other problems with how this is implemented.

I'm sorry if I'm coming across as demanding or ungrateful, I just have strong opinions on things sometimes. But I really appreciate the work you're putting in!

vhelin, 2022-02-05:
No worries. :) I think I'll open a new issue about this: .db {"label_index_%.2x", INDEX} ... We could reuse expand_variables_inside_string() here if the format was like {"label_index_%.2xINDEX"} (and the function would be enhanced so that it could handle that kind of syntax)...

vhelin, 2022-02-05:
Just made a commit where I added support for minimal formatting for substituted integers.

vhelin, 2022-02-06:
Ok, substitutions like Label_{COUNTER+1+BASE} work now. I also got rid of the special case for {\\@+1} as the substitutor uses now the stack calculator so {\\@+100-99} works now.

vhelin, 2022-02-09:
Label like

```

Label_{ \@ + 1 + 2 }:

```

should work now (previously spaces inside curcly brackets would make the parser fail with labels)...

vhelin, 2022-02-11:
@Drenn1 What do you think, is this issue solved or does it need more work?

Drenn1, 2022-02-11:
For some reason this doesn't work:

```
	.dw gfxHeader{5 - 1}
```

It gives this error:

```
build/data/gfxHeaders.s:2: INPUT_NUMBER: Got '' (0) when expected [0-9].
```

Also, attempting this:

```
.define COUNT 5

	.dw gfxHeader{COUNT - 1}
```

Gives the following error:

```
build/data/gfxHeaders.s:3: INPUT_NUMBER: Postponed calculation is not suitable for substitution as we need immediate results.
```

As far as I'm aware, both of these should work, right?

vhelin, 2022-02-12:
Found a reason why the 1st example didn't work and fixed that.

2nd example works for me, and the fact you are getting nondeterministic behavior scares me...

vhelin, 2022-02-12:
I think I found the reason for non-deterministic behavior: the stack calculator, when parsing (label) strings instead of source file, not all number parses could handle 0... I'll commit the fixes in a few minutes.

vhelin, 2022-02-12:
Uh, forgot to mention this issue in the commit, but if you now try the latest sources, do you still get problems?

Drenn1, 2022-02-16:
Sorry for the late reply, but that does seem to have fixed the issue.

There remains an edge case which bothers me, which is that this:

```
	.dw gfxHeader{AAAA}
```

Assembles successfully, even if ``AAAA`` is undefined, and is equivalent to ``.dw gfxHeaderAAAA``. I would expect it to attempt to resolve ``AAAA`` and fail. I'm not sure if this is intended, but it's sort of similar to what I was worried about before where the exact meaning of the syntax depends on specifically what was defined.

On the other hand this:

```
	.dw gfxHeader{AAAA + 0}
```

Gives an error as expected:

```
build/data/gfxHeaders.s:4: INPUT_NUMBER: Postponed calculation is not suitable for substitution as we need immediate results.
```

I would expect the above to happen even without the "+ 0".

As far as I can tell it is basically working though.

vhelin, 2022-02-16:
Good find! In your case

.dw gfxHeader{AAAA}

should not work. I'll fix the parser later today...

vhelin, 2022-02-16:
It seems I was too busy testing all that should work in a substitution that I forgot to test those cases that should not work. :)

Drenn1, 2022-02-16:
ok cool. Seems to work good now. :+1:

closed 2022-02-16 by Drenn1