vhelin/wla-dx #460: .IF comparison problems / segfaults

Drenn1, 2022-01-08:
Testing my project on the latest master branch, I seem to have run into a rather nasty bug with .IF comparisons. I know you're busy right now, but, whenever's good for you is fine. Unfortunately I have no idea how to handle stack.c which is the source of the issue.

I have code like this:

```
		.PRINT "VALUE: ", M_DELAY_INDEX, "\n"
		.if M_DELAY_INDEX != -1
			[...]
		.endif
```

Which produces this output:
```
VALUE: 6
include/script_commands.s:926: COMPUTE_STACK: Comparison between a string and a number doesn't work.
include/script_commands.s:926: ERROR: Couldn't parse ".if".
make[2]: *** [Makefile:144: build/ages.o] Segmentation fault (コアダンプしました)
```
Clearly, the value is not a string, but it claims I'm attempting a comparison between a string and an int. ...And then it segfaults.

I made a smaller test project. It didn't segfault this time, but the problem can still manifest itself. Valgrind is extremely enlightening here. When building the test project, it informs me that "Conditional jump or move depends on uninitialised value(s)", which to me is a very very bad sign, and probably the cause of the issue. It also shows the exact lines that are the likely culprits.

Test project code:

```
.define MY_VALUE -1

.if MY_VALUE != -1
	.PRINT "Value: ", MY_VALUE, "\n"
.endif
```

Below is my attempt to compile the attached test project. Notice how it prints "Value: -1" which it should not:
```
$ valgrind wla-65816 -i -o main.o main.s
==3049116== Memcheck, a memory error detector
==3049116== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==3049116== Using Valgrind-3.18.1 and LibVEX; rerun with -h for copyright info
==3049116== Command: wla-65816 -i -o main.o main.s
==3049116==
==3049116== Conditional jump or move depends on uninitialised value(s)
==3049116==    at 0x13A4CC: _comparing_a_string_with_a_number (stack.c:1541)
==3049116==    by 0x13A4CC: compute_stack (stack.c:1711)
==3049116==    by 0x13D315: stack_calculate (stack.c:1135)
==3049116==    by 0x117E0D: input_number (parse.c:366)
==3049116==    by 0x1242B5: parse_if_directive (pass_1.c:11002)
==3049116==    by 0x12CE12: parse_directive (pass_1.c:9478)
==3049116==    by 0x12F064: evaluate_token.part.0 (pass_1.c:1129)
==3049116==    by 0x130BB4: evaluate_token (pass_1.c:1124)
==3049116==    by 0x130BB4: pass_1 (pass_1.c:690)
==3049116==    by 0x1136EB: main (main.c:240)
==3049116==
==3049116== Conditional jump or move depends on uninitialised value(s)
==3049116==    at 0x13A4D5: compute_stack (stack.c:1713)
==3049116==    by 0x13D315: stack_calculate (stack.c:1135)
==3049116==    by 0x117E0D: input_number (parse.c:366)
==3049116==    by 0x1242B5: parse_if_directive (pass_1.c:11002)
==3049116==    by 0x12CE12: parse_directive (pass_1.c:9478)
==3049116==    by 0x12F064: evaluate_token.part.0 (pass_1.c:1129)
==3049116==    by 0x130BB4: evaluate_token (pass_1.c:1124)
==3049116==    by 0x130BB4: pass_1 (pass_1.c:690)
==3049116==    by 0x1136EB: main (main.c:240)
==3049116==
==3049116== Conditional jump or move depends on uninitialised value(s)
==3049116==    at 0x13A4EE: compute_stack (stack.c:1720)
==3049116==    by 0x13D315: stack_calculate (stack.c:1135)
==3049116==    by 0x117E0D: input_number (parse.c:366)
==3049116==    by 0x1242B5: parse_if_directive (pass_1.c:11002)
==3049116==    by 0x12CE12: parse_directive (pass_1.c:9478)
==3049116==    by 0x12F064: evaluate_token.part.0 (pass_1.c:1129)
==3049116==    by 0x130BB4: evaluate_token (pass_1.c:1124)
==3049116==    by 0x130BB4: pass_1 (pass_1.c:690)
==3049116==    by 0x1136EB: main (main.c:240)
==3049116==
Value: -1
DISCARD: main.s: Discarding an empty section "Bank0".
==3049116==
==3049116== HEAP SUMMARY:
==3049116==     in use at exit: 0 bytes in 0 blocks
==3049116==   total heap usage: 294 allocs, 294 frees, 325,762 bytes allocated
==3049116==
==3049116== All heap blocks were freed -- no leaks are possible
==3049116==
==3049116== Use --track-origins=yes to see where uninitialised values come from
==3049116== For lists of detected and suppressed errors, rerun with: -s
==3049116== ERROR SUMMARY: 3 errors from 3 contexts (suppressed: 0 from 0)
```

Testing on commit a9ba720db0ac810b2046503006db6569550db3f8.

Test case: [testcase.zip](https://github.com/vhelin/wla-dx/files/7832403/testcase.zip)

vhelin, 2022-01-08:
Thanks for reporting this, and yeah I think I have time to check this out at the end of next week... But I'll try to fix this asap.

vhelin, 2022-01-16:
The funny thing is that I cannot get that code to crash under Cygwin/Windows 10 and Ubuntu 20.04 Linux box. Valgrind on the Linux box reports things like you show here, and the .IF doesn't work properly... Let me see...

vhelin, 2022-01-16:
The stack calculator failed to parse that -1 after != properly, or it was two items, minus sign and 1. Before != on that like it was able to merge those into -1. Now it should work.

Drenn1, 2022-01-16:
Awesome, I'll give this a test drive tomorrow.

Drenn1, 2022-01-17:
This _appears_ to be resolved, though it's hard for me to say for sure until I fix the remaining problems causing my project's compilation to fail. I have at least two more bugs to look into :upside_down_face: 

I wish I had kept tabs on the status of my project's compatibility with newer WLA versions, it would have made both of our lives easier!

Drenn1, 2022-01-20:
Since my project now compiles correctly I consider this resolved!

closed 2022-01-20 by Drenn1