vhelin/wla-dx #462: Segfault on exit when freeing hashmaps

Drenn1, 2022-01-18:
When attempting to assemble my project (https://github.com/drenn1/oracles-disasm) it crashes after it's been assembled, when it's attempting to free the hashmaps. (This is AFTER it produces the output, so I do actually get my output file).

I haven't been able to pin down the problem yet or make a smaller testcase. But, I ran valgrind again, and everything points to the hashmap_free function. Specifically the call on line 520 of main.c, ``hashmap_free(s1->nspace->label_map);``. It looks to me like ``nspace->label_map`` isn't a valid pointer somehow, as indicated by the invalid read in hashmap.c:350 (``m->table_size``?). I'm not sure why this would be the case though.

Valgrind output: https://gist.github.com/Drenn1/80c62fc85125522455ef03b42d5ee65d

I'll provide more details, such as a proper testcase if/when I can produce them.

vhelin, 2022-01-19:
Hashmaps in WLA DX are a black hole for me as I didn't implement them, or made the old code to use them. :) I think I really need a testcase that makes WLA to crash to be able to fix this issue...

vhelin, 2022-01-23:
I just spent yesterday evening playing with hashmaps in WLA, and can say that this error

```
Invalid read of size 4
==313209==    at 0x114177: hashmap_free (hashmap.c:350)
```

... comes when you try to give hashmap_free a NULL pointer. Don't know about the other errors, but yeah, a testcase would be nice. :)

cr1901, 2022-01-23:
I wonder how difficult it would be to add functionality to `ctest` to run the freshly compiled binaries against a "GH test cases" directory, like LLVM does. Test cases of bugs opened on GH, old and new, would go into it.

I think this is what the `bug_exhibition` directory was for, but people (incl me :P) don't seem to want to make PRs to add them...

Drenn1, 2022-01-25:
^ There is the "tests" directory already, do you mean something different? Ville seems to be often updating the testcases to check for specific issues as they come up.

Anyway, success! I got a testcase. This one actually isn't supposed to assemble. Regardless, it segfaults, perhaps for the same reason as my project. This is just a minor change to "gb-z80/namespace_test", with another namespace added at the end to trigger the bug.
[namespace_test.zip](https://github.com/vhelin/wla-dx/files/7937607/namespace_test.zip)

Drenn1, 2022-01-25:
My theory: The fact that multiple sections can share the same namespace is causing a double-free. The problem would not have manifested in namespace_test earlier because, although it does contain two sections with the same namespace, they are not assembled into the same object file.

vhelin, 2022-02-03:
I'll test this more, but hopefully my small fix solves this issue. :)

vhelin, 2022-02-03:
Found a potential memory leak there. Anyway, if you test now with the latest sources, is the issue gone?

Drenn1, 2022-02-05:
The issue is gone, thanks! I'd recommend making a new release sometime soon. Some of those bugs were fairly substantial, and any version of WLA that can build my monstrosity is probably in good shape.

closed 2022-02-05 by Drenn1

vhelin, 2022-02-05:
I'm also thinking that I should make a release soon, but the definition substitution feature I just added (https://github.com/vhelin/wla-dx/issues/459) needs some testing to be done, and comments, before I can release v10.1... :)