vhelin/wla-dx #463: CADDR not working properly

Drenn1, 2022-01-18:
Despite issue #462 I am able to produce object files with wla-gb, so it's on to the next issue preventing my project from assembling.

I have the following code which takes advantage of "CADDR" to define a relative address which can fit in a byte, instead of needing to use a word:

```
@spritesTable:
	.db @sprites0-CADDR
	.db @sprites1-CADDR
	.db @sprites2-CADDR

@sprites0:
	.db $08
	.db $08 $48 $04 $88
	.db $18 $48 $04 $88
	.db $08 $50 $04 $88
	.db $18 $50 $04 $88
	.db $08 $68 $04 $88
	.db $18 $68 $04 $88
	.db $08 $70 $04 $88
	.db $18 $70 $04 $88

@sprites1:
	.db $02
	.db $08 $30 $04 $88
	.db $18 $30 $04 $88

@sprites2:
	.db $06
	.db $08 $28 $04 $88
	.db $18 $28 $04 $88
	.db $08 $88 $04 $88
	.db $18 $88 $04 $88
	.db $08 $90 $04 $88
	.db $18 $90 $04 $88
```

This used to work, but now this gives the following output:

```build/ages.o: code/bank2.s:6186: COMPUTE_PENDING_CALCULATIONS: Result (-20682/$ffffaf36) of a computation is out of 8-bit range.```

This is clearly wrong, it seems to have somehow become a negative number despite the fact that the labels it is referencing should have higher addresses than CADDR, thus the subtraction should produce a small positive number.

I'll give more details such as a proper testcase when I can produce it.

Drenn1, 2022-01-18:
By the way, although unrelated to this issue, this is not to mention that operator precedence seems to have changed and I needed to change my code to accomodate it. Since this was mentioned in the changelog I suppose this was intentional. Specifically I had to change many uses of the "<<" operator whose precedence is now, as the changelog said, more C-like. This is fine, just wanted to give you a heads-up about how that change affected compatibility.

vhelin, 2022-01-18:
That's why I upped the major number of the release where operator precedence was changed, to indicate that it contained some big changes. :) Also, like you did, reading the CHANGELOG helps. And yeah, it was intentional.

vhelin, 2022-01-18:
Hmm, seems like the CADDR test in the tests folder doesn't check the validity of the assembled binary, it only checks that the assembler can assemble sms.s... I think I'll try to make the test use byte_tester...

vhelin, 2022-01-18:
As a test I assembled your test code and it assembled just fine and the result was what was expected... ? Do you have the latest sources, and made "make clean" + "make"?

Drenn1, 2022-01-19:
I'm definitely using the latest version. My project is enormous so there must be some combination of things going on to cause this problem - similar to how issue #462 doesn't appear to affect any of the test cases. The hard part for me will be making a test case to isolate the problem!

Drenn1, 2022-01-19:
I no longer think this is a problem with CADDR. It must be a problem with child label name resolution.

I added the following code to the end of tests/z80/caddr_test:

```
;------------------------------------------------------------------------------
; Yet another CADDR test
;------------------------------------------------------------------------------

.section "TEST-03" free keep

mylabel:
	.db @sprites0-CADDR

@sprites0:

.ends
```

This fails to compile, but ONLY when the label is named ``@sprites0`` or some other child label name that was previously defined. Give it any other name, and it will assemble (though I haven't checked its correctness).

vhelin, 2022-01-19:
Haha, I failed to remember that those @labels were child labels and thus gave them unique names in my tests thus avoiding this situation. :)

Anyway, thanks, I was able to repeat that weirdness:

```
sms.o: sms.s:174: COMPUTE_PENDING_CALCULATIONS: Result (969/$3c9) of a computation is out of 8-bit range.
```

I bet the new feature added in v10.1a still has issues with child labels:

```
[ALL] The assembler is now able to calculate A-B where A and B are labels in the same section. Only the linker used to be able to do this.
```

I remember fixing one child label related issue that this new feature caused, so this could be another one... I'll try to fix this later today.

vhelin, 2022-01-19:
Ok, this was a child label related issue. Hopefully my fix works, I added more tests to test child labels...

vhelin, 2022-01-19:
Hmm something is wrong. tests/gb-z80/delta_test worked under Cygwin, but fails under Linux. Need to investigate.

vhelin, 2022-01-19:
Ah, I forgot to make all WLA executables before testing. But found out that the test itself was broken, fixed that.

Drenn1, 2022-01-20:
OK great - this does appear to fix the issue. My project does assemble correctly now! But there are still some issue remaining. I will of course need to look into issue #462, find the reason for slower compilation (running wla-gb takes 43 seconds instead of 5 or less), and another issue that I will open in a moment.

closed 2022-01-20 by Drenn1