vhelin/wla-dx #464: Muting warnings for cross-bank calculations

Drenn1, 2022-01-20:
Seems related to issue #314.

My project now assembles after the latest round of fixes, but running wlalink produces 1202 lines of output in the console that look like this:

```
build/audio.o: include/musicMacros.s:130: COMPUTE_STACK: The passed on BANK changed from $3e to $39. This might have no effect, but just to let you know. Please check that the result of this calculation is correct.
build/audio.o: include/musicMacros.s:130: COMPUTE_STACK: The passed on BANK changed from $3e to $39. This might have no effect, but just to let you know. Please check that the result of this calculation is correct.
build/audio.o: include/musicMacros.s:130: COMPUTE_STACK: The passed on BANK changed from $3e to $39. This might have no effect, but just to let you know. Please check that the result of this calculation is correct.
build/audio.o: include/musicMacros.s:130: COMPUTE_STACK: The passed on BANK changed from $3e to $39. This might have no effect, but just to let you know. Please check that the result of this calculation is correct.
```

The following line of code is responsible - I need to calculate the difference between two bank numbers:

```
.db :\1Start - :b39_initSound ; Bank number
```

Similar to issue #314, this code worked perfectly fine, and continues to work fine, but now is producing warnings for some reason. I don't see what the problem could be with performing arithmetic on bank numbers.

Perhaps the warning isn't meant for this case; perhaps it's only meant for when you perform arithmetic on addresses (rather than bank numbers) that don't share the same bank. But even then I don't really understand why the warning is necessary?

As an aside, thanks for your diligence with the fixes you've applied lately, it is appreciated!

vhelin, 2022-01-20:
1202 lines of the same warning is a bit too much, especially if the warning is kinda pointless. :) I think that when I added that warning I had some special calculation in mind where that warning would make sense, but I cannot remember what that was. 

Looking at the code I see similar warnings for BASE and SLOT thus it might be that I just extrapolated a potential problem in my mind to happen also with BANK, not sure.

I'm thinking that I should add a flag like -V and put all those "be extra careful now" warnings behind it. All ideas are welcome. Now it's getting late here, I'll try to fix this issue tomorrow one way or another.

Drenn1, 2022-01-20:
I thought at first when I opened issue #314 that this was a warning informing the user that they were doing something they probably didn't mean to do; but your response suggested that there was actually a technical reason for it. I just have trouble seeing it, though. At the end of the day, these are all just numbers, and I don't understand why there would be any limitations on the types of arithmetic we can perform on them.

It all depends on why the warnings were added, I guess, which is what I don't understand. If it's for the user's convenience, then, I guess put it behind a "verbose" flag.

If you deem that these warnings _are_ important - well, I don't know what would be best, it would kind of depend on what the reason for the warnings is. Like, if there's actually some edge-case where that code _won't_ work then maybe I should be concerned about it.

vhelin, 2022-01-20:
I'm thinking about putting them behind -V (-v is the verbose flag) that would be like extra special verbose. Perhaps -w would be better? Something that one could use if the assembled thing wasn't working and perhaps -w could give some insight into it? Anyway, a new flag.

PS. I'll try to optimize stack.c:data_stream_parser_parse() this weekend, should not be that hard and that should give a nice speedup for files with lots of calculations... ? And I think the very last test case is quite slow on my computer as well, I should profile it and see how to make it faster...

lhsazevedo, 2022-01-20:
Some tools use -v, -vv and -vvv

Drenn1, 2022-01-20:
What I really want to know is what the warning actually _means_. And if it only applies to 1% of cases, what is that 1%? Can it be adjusted to only print in those more limited circumstances - or even to cause an error in that case? Normally if a calculation can't be computed correctly then it should fail to assemble. I feel that getting to the root of the problem would be preferable to hiding this behind verbose compilation flags.

Drenn1, 2022-01-20:
And of course, anything that can improve performance would be appreciated! But I think there must have been a specific regression between v9.11 and now that caused compilation time to increase several-fold. For example, is the new code that handles relative label arithmetic in the assembler using hashmaps? If not - I suspect that would be problem.

vhelin, 2022-01-21:
>What I really want to know is what the warning actually means.

I think it means what it says. :) "Are you sure you want to do this?" The more I think about this the more I'm sure that I was overreacting when I wrote that warning and thus I'll be taking it away. The user should know what he is doing when he writes those calculations...

vhelin, 2022-01-23:
> And of course, anything that can improve performance would be appreciated! But I think there must have been a specific regression between v9.11 and now that caused compilation time to increase [...]

If you now try with the latest sources, has the speed improved? I made three nice optimizations to WLA, one to old code and two to new code in v10.1a...

Drenn1, 2022-01-25:
I appreciate the work on the optimizations. It is indeed much faster now - about 2.6 seconds now! But I'm now getting this error:

```
include/rominfo.s:12: ERROR: Unknown symbol "$4000".
```

Referring to this line:

```
.banksize $4000
```

Oddly, it works if I change id to ".rombanksize". Apparently these two opcodes are supposed to do the same thing. I had no idea I was using an old alias of it.

The commit at fault appears to be 61d007f8a9b1fc988997d4af33cdbf604ae954c3. Was it removed on purpose? According to a quick glance at the code, it seems like it's still supposed to work.

I will try to create a testcase for the hashmap issue tomorrow...

vhelin, 2022-01-25:
Sorry about .BANKSIZE gone missing, it happened when I optimized parse_directive()...

Drenn1, 2022-01-25:
Cool, seems fixed now. We've strayed pretty far from the original purpose of this issue, but that one's fixed too. =)

closed 2022-01-25 by Drenn1