vhelin/wla-dx #465: Unresolved reference with _sizeof in certain cases

aexoden, 2022-01-23:
I've tried to make this as minimal a test case as I can, but it seems to require a slightly more complex setup to trigger. I can say that it appears to start with commit 2ee4ea14fed47b7abeb87610601eded57d2bd8e4. I had to separate the files to get the error to trigger, but maybe I didn't get the correct set of conditions in any of my single file tests. 

Given the following files:

test.s
```
.include "test.inc"

main:
	LDA test_var1 + _sizeof_test_struct
	LDA test_var1 - _sizeof_test_struct
```

test.inc
```
.memorymap
	slotsize $8000
        defaultslot 1
        slot 0 $0000
        slot 1 $8000
.endme

.lorom

.rombanksize $8000
.rombanks 1

.struct test_struct
	var1 db
	var2 db
.endst
```

ram.s
```
.include "test.inc"

.base $7E
.ramsection "RAM" bank 0 slot 0
	test_var1 db
.ends
```
Linkfile
```
[objects]
ram.o
test.o
```

After building the two .s files with wla-65816 and then attempting to wlalink, I get the following error:

```
test.o: test.s:5: PARSE_STACK: Unresolved reference to "_sizeof_test_struct".
```

Note that this refers to line 5. The addition seems to work fine, but the subtraction triggers the error.

vhelin, 2022-01-23:
It was related to this newish feature:

"[ALL] The assembler is now able to calculate A-B where A and B are labels in the same section. Only the linker used to be able to do this."

I hope it works again now. :)

aexoden, 2022-01-23:
It does indeed appear to be working with the latest updates. Thanks.

closed 2022-01-23 by aexoden