vhelin/wla-dx #466: Substitute value of .DEFINE'd variable into a string inside quotation marks

vhelin, 2022-02-05:
This feature would be much like the one implemented in https://github.com/vhelin/wla-dx/issues/459 only this time it'd affect strings inside quotation marks.

Perhaps we could also add string formatting to the old parser?

Perhaps something like:

.db {"label_index_%.2x", INDEX}

or

.db {"label_index_%.2x{INDEX}"}

?

vhelin, 2022-02-05:
Took the

.db {"label_index_{%.2x{INDEX}}"}

route as there we could enhance and reuse parse.c:expand_variables_inside_string().

closed 2022-02-05 by vhelin

Drenn1, 2022-02-06:
Interesting. That solves the problem of string substitution I suppose. I'm not sure if you wrote tests for this, but I also tested the format strings on labels (instead of strings) and it does seem to work.

I know I suggested this, but keep in mind that this might make issue #467 unworkable. The contents of the curly brackets are becoming their own syntax, instead of something simple that could be replaced with the stack parser in the same way as any other calculation.

Maybe consider alternatively using a ``.definef`` opcode which would have the advantage of working in a more general way and not interfere with the current simplicity of the bracket substitution. Though it could be more wordy depending on the application.

If you're going to keep this anyway, could you add documentation on the format strings to the documentation?

vhelin, 2022-02-06:
> Interesting. That solves the problem of string substitution I suppose. I'm not sure if you wrote tests for this, but I also tested the format strings on labels (instead of strings) and it does seem to work.

I added some tests to tests/z80/substitution_test for this.

> I know I suggested this, but keep in mind that this might make issue https://github.com/vhelin/wla-dx/issues/467 unworkable. The contents of the curly brackets are becoming their own syntax, instead [...]

I'm thinking that we can still detect calculations by lookings for symbols like + and - between curly brackets, so 467 should still be doable.

> Maybe consider alternatively using a .definef opcode (as I floated in https://github.com/vhelin/wla-dx/issues/459) which would have the advantage of working in a more general way and not interfere with the current simplicity of the bracket substitution. ...Although that's not exactly ideal either because a .definef directive would only work on strings, not labels.

I don't see the reason for adding .definef

> If you're going to keep this anyway, could you add documentation on the format strings to the documentation?

I wrote about this in doc/asmsyntax.rst

Drenn1, 2022-02-06:
> I'm thinking that we can still detect calculations by lookings for symbols like + and - between curly brackets, so 467 should still be doable.

Sorry to say this again, but this solution rubs me the wrong way - in a similar way as I didn't like ``.db "HELLO_{COUNT}"`` working if ``COUNT`` is undefined, because the meaning of exactly what's going on changes depending on context. This isn't as bad as that, but I would imagine that any code written to detect whether arithmetic is happening this would probably have at least a few difficult to detect edge-cases.

Here's a weird edge-case for you:

```
label1:
  .db $01
label2:
  .db $01

.db {"Label1 size: {_sizeof_label1}"}     ; Fails (not a define)
.db {"Label1 size: {_sizeof_label1 + 0}"} ; Succeeds (stack parser can handle it)
```

In fact, if we assume that the stack parser would be flexible enough to handle link-time name resolution but the curly bracket substitution code is not, then the problem actually gets much worse because this would apply to any label or exported definition.

If we assume that you can work around the above problem (detecting the 1st case as one in which to use the "stack parser"), this problem below would still be quite confusing, and I think not possible to work around, which I think demonstrates the uncomfortableness I have with context-sensitivity here.

```
.db {"Label1 size: {%.2x{_sizeof_label1}}"} ; Fails (not a define)
.db {"Label1 size: {_sizeof_label1}"}       ; Succeeds (stack parser can handle it)
```

> I don't see the reason for adding .definef

It's just an alternative suggestion. It would keep the syntax of the curly braces more elegant.

```
.definef MY_LABEL_NAME "label_%.2x", INDEX
.dw {MY_LABEL_NAME}
```

But I'm not suggesting to add it in addition to the current syntax,  just as a potential replacement - because indeed I don't see much need for the ability to define strings like that beyond this use case.

> > If you're going to keep this anyway, could you add documentation on the format strings to the documentation?
> 
> I wrote about this in doc/asmsyntax.rst

Yeah dunno how I missed that.

---

vhelin, 2022-02-06:
About your edge case:

.db {"Label1 size: {_sizeof_label1}"}     ; Fails (not a define)

_sizeof_label1 is actually a definition so that doesn't fail, if the definition exists at the time we process that string. And same here:

.db {"Label1 size: {_sizeof_label1 + 0}"} ; Succeeds (stack parser can handle it)

That would succeed only if _sizeof_label1 existed at the point of processing of that string. There is no mechanism in WLA currently that could make substitutor to postpone the processing of that string to WLALINK. The substitution function would need an immediate result from the stack calculator or it would fail.

> It's just an alternative suggestion. It would keep the syntax of the curly braces more elegant.

I think the current solution is more elegant. :D Perhaps code "elegance" is in the eye of the beholder...

Drenn1, 2022-02-06:
"_sizeof" was a bad example I think - because it _should_ work - but you might want to double-check that because it's not working for me when used in curly brackets! Are you sure it's not defined in the linker?

The more important point is that the stack parser will most likely be able to handle any exported definitions or labels while the normal curly-brace substitution will not. At the very least this will be a gotcha to look out for.

> I think the current solution is more elegant. :D Perhaps code "elegance" is in the eye of the beholder...

I mean, yeah the way I presented it it probably doesn't look that elegant. But my vision was this: The contents of the curly brackets get evaluated in the same way as any other calculation in WLA, then have their value substituted in. To me, that would be elegant. But it is incompatible with the format strings thing - or at least it will make your life a lot harder to support both and there will probably be a bunch of edge-cases - hence my suggestion that it be done with another mechanism.

vhelin, 2022-02-06:
This works:

```
        .RAMSECTION "RAMSection" BANK 0 SLOT 0
label1: db
label2: dsb 2
        .ENDS

        .PRINT "_sizeof_label1 = "
        .PRINTV DEC _sizeof_label1
        .PRINT "\n"

        .DB "04>"                    ; @BT TEST-04 04 START
        .DB {"{_sizeof_label1}"}     ; @BT 31
        .DB "<04"                    ; @BT END

```

If you want to test if {_sizeof_label1 + 0} would work in the same place, do something like

        .PRINTV DEC (_sizeof_label1+0)

as .PRINTV also needs immediate result from the stack calculator.

> The more important point is that the stack parser will most likely be able to handle any exported definitions or labels while the normal curly-brace substitution will not. At the very least this will be a gotcha to look out for.

It wouldn't work like that. Curly brace substitution can only work with definitions it knows, and if it'll use stack calculator in the future then the same limitation applies to stack calculator in that case.

vhelin, 2022-02-06:
This boils down to the age old issue that WLA cannot do all things WLALINK can do as WLA doesn't know what other object files contain, only WLALINK does know that, and some things WLA needs to do right away, cannot postpone those to WLALINK.

Then again you could just assemble everything into one object file using .INCLUDEs and "link" that, then WLA would have access to all definitions your project has. As long as they get defined before the curly brace substitutor tries to access the definitions. :)

vhelin, 2022-02-06:
One thing to make postponing work here: force the user to specify the number of bytes each substitution generates in a string thus WLA would know the number of bytes the substitution occupies, allocate those, and WLALINK could just overwrite those bytes later.

Drenn1, 2022-02-06:
> If you want to test if {_sizeof_label1 + 0} would work in the same place, do something like
> 
> ```
>     .PRINTV DEC (_sizeof_label1+0)
> ```
> 
> as .PRINTV also needs immediate result from the stack calculator.

It is working with ramsections but not with labels from normal sections. Even though the "_sizeof" labels appear in the [definitions] section of the .sym file.

> It wouldn't work like that. Curly brace substitution can only work with definitions it knows, and if it'll use stack calculator in the future then the same limitation applies to stack calculator in that case. The important thing here is that WLA is doing the curly brace substitution, not WLALINK.

My bad. I did not read your previous reply closely enough.

So yeah maybe this can be workable. Just, rather than searching for arithmetic operators maybe it would make sense to check if "%" is the first character in the substitution. ...That would almost work except for binary numbers (%0101"). But it would still be much better than seacrhing for arithmetic operators in my opinion. Having something very clear to check for at the beginning of the curly brace, which explicitly states "this is not arithmetic, this is string substitution", should be good enough to differentiate the two cases.

So, this is string formatting, because it detects "%.2x" at the start:
``label_{%.2x{INDEX}}``
And this is something to be handled by the stack parser because the format string is not detected:
``label_{INDEX}``

If this will work to differentiate the two cases - rather than checking for arithmetic operators in use - then I think that would alleviate my concerns.

Drenn1, 2022-02-06:
I think I am done for today, sorry for bugging you so much about this. You clearly know what you're doing, I think I was just getting worked up over minor, solvable details.

vhelin, 2022-02-06:
> It is working with ramsections but not with labels from normal sections. Even though the "_sizeof" labels appear in the [definitions] section of the .sym file.

Then WLA doesn't know those and WLALINK possibly generates those later... Which is too late.

...

Oof, there is a problem: in the case of postponing a substitution to WLALINK the WLA's code that expects a string from the parser would get a string with padding and then it would need to check if there is a postponed reference waiting, and then place that string and the reference into the output stream - two things instead of just one. And there are quite a many places where WLA's code expects a string... I need to think more about this, might be too hard to implement.

...

> So yeah maybe this can be workable. Just, rather than searching for arithmetic operators maybe it would make sense to check if "%" is the first character in the substitution. ...That would almost [...]

I don't know why you are against searching for arithmetic operators as parse.c:input_number() has been doing that for decades... :)

vhelin, 2022-02-06:
Oh yeah, consider this example

.db { "{%.2X{DEF_A}}{%.2X{DEF_B}}{%.2X{DEF_B}}" }

If WLA didn't know any of those definitions it'd create a string "000000" and it would have to deliver three postponed substitution structs to the code that is now expecting just a string or perhaps an integer...

Postponing substitutions to WLALINK quickly explodes into a heap of problems so I'm thinking that we'll have to live what we now have - WLA needs to do the substitution or it fails.

Drenn1, 2022-02-06:
Yes that's fine, I don't think this feature needs to involve the linker upon further consideration.

> I don't know why you are against searching for arithmetic operators as parse.c:input_number() has been doing that for decades... :)

It's the idea of searching for arithmetic operators to determine the overall purpose of the curly brackets that bugs me. Using their existence to determine the context. I feel that if the curly brackets can do two different things then it's best to have explicit syntax that determine which of those things it's doing at any given moment. Which, it kind of does, so that syntax should be used to determine the context. In fact, ideally the same code that checks for "%.2s" for the format string implementation would also be the code that determines "ok, so these curly brackets are NOT being used for arithmetic". Because that would be more maintainable and less likely to contain bugs.

Drenn1, 2022-02-06:
To put it another way: I believe that the use of the arithmetic / stack parser or whatever it's called should be the rule, not the exception, and format strings should be treated as a special-case.