vhelin/wla-dx #468: Increase or remove MAX_STRING_LENGTH

maxim-zhao, 2022-02-08:
Now that we have line continuation, I can try to do things like this:
```
.stringmap menus \
"┌── MENU ──╖$" \
"│  EXIT    ║$" \
"│  COLOR   ║$" \
"│  ERASE   ║$" \
"│  SQUARE  ║$" \
"│  CIRCLE  ║$" \
"│  ELLIPSE ║$" \
"│  PAINT   ║$" \
"│  COPY    ║$" \
"│  MIRROR  ║$" \
"│  MAGNIFY ║$" \
"│  DISPLAY ║$" \
"│  END     ║$" \
"╘══════════╝$"
```
Except... 
```
smsgraphicboard.sms.asm:1321: WARNING: String parsing was interrupted due to buffer getting full.
smsgraphicboard.sms.asm:1321: INPUT_NUMBER: The string is too long (max 255 characters allowed).
smsgraphicboard.sms.asm:1321: DIRECTIVE_ERROR: STRINGMAP: no string givensmsgraphicboard.sms.asm:1321: ERROR: Couldn't parse ".stringmap".
```
It's 2022, can we make this limit go away, or be runtime-configurable, or at least be a few KB?

vhelin, 2022-02-08:
Runtime configurable: not really as in a lot of places we allocate char buffers from stack using MAX_NAME_LENGTH.

Few KBs is too much for Amiga, I think, but I could make it so that on Amiga it defaults to the current 255 and on other platforms it's 1024?

vhelin, 2022-02-08:
I think I'll up that to 1KB and then before committing it test how it works on Amiga. If it works then I'll use 1KB for all platforms. But before that I need to finish my workday at the office... :)

vhelin, 2022-02-08:
Can't grow the value to 2K or more at the moment as I know there are some places where we combine multiple strings into a 4K buffer and that code might break if the source strings are 2K+... I'll need to revisit that code before making MAX_NAME_LENGTH bigger than 1023...

maxim-zhao, 2022-02-08:
I guess the correct thing to do is to make some helper methods to reallocate and free as needed - static stack buffers are an easy hack to get around the pain. I may offer some PRs around this subject...

vhelin, 2022-02-08:
There are 130 references to MAX_NAME_LENGTH in WLA's c files and 25 in h files so there should be plenty of fun for you for some time. :)

vhelin, 2022-02-09:
Ok, MAX_NAME_LENGTH is now 2047 unless talking about Amiga or MSDOS where it'll be 511. I hope this solves this issue for now...

maxim-zhao, 2022-02-10:
> There are 130 references to MAX_NAME_LENGTH in WLA's c files and 25 in h files so there should be plenty of fun for you for some time. :)

There's plenty more things reliant on it too...

closed 2022-02-10 by maxim-zhao