vhelin/wla-dx #483: `RAMSECTION` does not support `SIZE` property

Kroc, 2022-02-19:
Is this a limitation or an oversight? You cannot specify a size-limit for a `RAMSECTION`. This is useful for hardening code against accidental overflow mistakes and also when you need to enforce that a table fit into 256 bytes, for example.

The `SIZE` property should also apply to `[ramsections]` in link files.

I don't know what the current behaviour is regarding `APPENDTO` sections also having a `SIZE` property; how would this behave with `RAMSECTION`s?

vhelin, 2022-02-25:
SIZE is missing from .RAMSECTION just because I didn't think it'd have any use there. After reading your issue I think we should add it there... :)

If I read WLALINK's code correctly, if you have section B which is APPENDTO to section A, then section A's size will become A's size + B's size when the merging is done. I'm suggesting we do the same with .RAMSECTIONs...

Kroc, 2022-02-25:
Thanks! Appending sizes sounds like the right thing to do; that way optional includes could expand a RAM section and the `size_of` values et al will still be usable elsewhere in the code.

vhelin, 2022-02-25:
After reading the initial issue description I have to say that currently the feature doesn't give errors/warnings about overflows. Forgot this requirement. :) Will make a new commit soon...

vhelin, 2022-02-25:
Oops, the code should work. The code was better than what I thought it would be. :)

vhelin, 2022-02-26:
I have one question: what if the user uses SIZE in [ramsection] and gives a value that's smaller than the section's current size is? Should this give an error?

Kroc, 2022-02-26:
Yes, absolutely, because they may be assembling a game for two different platforms that have different sizes of RAM and the code may be the same, but the linker config will differ.

vhelin, 2022-03-08:
Can this issue be closed now? Is that SIZE added to .RAMSECTION working as it should? I'm thinking yes, but a confirmation would be nice... :)

Kroc, 2022-03-18:
Sorry for the delay, yes I've rebuilt and tested and it works -- although it only issues a warning rather than a failure, but the semantics of that are a separate issue

closed 2022-03-18 by Kroc

vhelin, 2022-03-19:
Just tested it myself and it seems that if you try to shrink a section in linkfile then that'll give an error, but if you shrink the section in the assembler then that gives a warning (and doesn't shrink it)...

I guess I'll turn that assembler warning into an error...