vhelin/wla-dx #489: Paths in WLALINK's linkfile are relative the CWD, not where linkfile is

vhelin, 2022-03-02:
### Discussed in https://github.com/vhelin/wla-dx/discussions/487

<div type='discussions-op-text'>

<sup>Originally posted by **Pokeman2003** March  2, 2022</sup>
Hi all! I posted to SMSPower about this question earlier, but the only one so far to offer help, maxim, doesn't know how to fix it either. Long story short, I have a batch file automating my project compilation. It looks a bit like this.
```batch
del "finloc"\object.o
del "finloc"\listfile
"compilerlocation\wla-z80.exe" -o "projectlocation"\project.asm "finloc"\object.o
echo [objects]>"finloc"\listfile
echo object.o>>"finloc"\listfile
"compilerlocation\wlalinker.exe" -d -r -v -s "finloc"\listfile "finloc"\output.sms
```
Compilerlocation, finloc, and projectlocation are all relative paths, and thus have quotation marks around for me to avoid worrying about a space somewhere.

Now, this code seems like it SHOULD work, but when I run it, the linker returns an error, complaining to me that it could not open `object.o`. The file DOES exist in the intended location, so I'm not entirely sure what's going wrong. I tried putting `"finloc"\object.o` in the listfile, but then it just complains it can't open "finloc", and -L switch did nothing to change my situation.

It was suggested that wlalinker worked with relativity, which would potentially make sense.  So the question is, how do I get WLALinker to operate in other folders that it's not in?</div>

SOLUTION:

Add a flag to WLALINK that'd make paths in linkfile to be relative to its location

vhelin, 2022-03-02:
I'll try to add that flag to WLALINK this week, perhaps tomorrow?

Pokeman2003, 2022-03-02:
I can confirm it works on the new binaries!
Except, strangely, it even works on my original wlalinker. Was this a switch I just missed entirely when skimming over the documentation?

vhelin, 2022-03-02:
Hmm perhaps you somehow mixed up files? There was no -R switch in the old WLALINK, I just added it like 5 hours ago. :)

closed 2022-03-08 by vhelin