vhelin/wla-dx #490: Linker doesn't pipe output with Command Prompt

Pokeman2003, 2022-03-02:
cmd.exe has the ability to output STDOUT into a file. You do this by doing something akin to `command -arguments>>fileyouwantoutput.log`.  WLALink doesn't appear to support this, instead forcing its output on the screen.
I don't think the assemblers do either, but they don't output anything unless there's an error, as far as I'm aware.

vhelin, 2022-03-02:
Looking at the code I can say that WLALINK outputs at least the summaries to STDERR. Which is kinda wrong, I think. Perhaps just a copy paste error when writing that code. Errors should always go to STDERR.

I think I'll fix the summary issue in WLALINK right away...

vhelin, 2022-03-02:
The assemblers also printed their summaries to STDERR, a fix is coming up soon... Otherwise warnings and errors should go to STDERR, the rest to STDOUT.

vhelin, 2022-03-02:
Ok, with the latest sources the problem should be fixed. You might also want to look also how to forward STDERR to a text file. On some shells it's "2>output.txt"...

Also on some shells "2>&1" directs STDERR to STDOUT...

Pokeman2003, 2022-03-02:
It'd be `2>fileyouwantoutput.log` for cmd.exe. Thank you for directing me! I wasn't aware Windows' Command Prompt even let you do that.

maxim-zhao, 2022-03-04:
All the above redirectors work identically in Windows cmd shell as in Bash in Linux: `2>file`, `1>file`, `2>&1`. However some build systems may treat stderr output as errors so it's best to output only errors there.

vhelin, 2022-03-08:
I think this is now fixed...

closed 2022-03-08 by vhelin