vhelin/wla-dx #491: FOPEN causing problems with -M option

rondnelson99, 2022-03-04:
I've written a macro that includes a binary file while rearranging all the bytes. It looks like this:
```
.MACRO incv2bpp ARGS FILE WIDTH ;include a 2bpp row-major image, but convert to column-major
	.fopen FILE HANDLE
	.fsize HANDLE SIZE
	.REPT WIDTH/8 INDEX X
		.REPT SIZE/(WIDTH/8) INDEX Y
			.fseek HANDLE (WIDTH/8*Y)+X START
			.ftell HANDLE LOC
			.printv LOC
			.print "\n"
			.fread HANDLE DATA 
			.db DATA
		.ENDR
	.ENDR
.ENDM
```
I've also adapted the Makefile from ISSOtm's gb-starter-kit (intended for RGBDS), and am using WLA's `-M` option to automate resource generation. If I use .INCBIN on a file, when WLA is run with -M and the file does not exist, it emits no errors, but tells `make` to generate the included file according to its rules. I would expect `.fopen` to act the same way when the file does not exist, but it seems to error out and halt the build process. It's true that I'm far from an expert with `make`, so that could be the source of my problems, but nonetheless it seems there is some sort of discrepancy between `.incbin` and `.fopen`. Is there a reason for this?
Thanks!

rondnelson99, 2022-03-04:
I'm aware that this example could totally be implemented using `.incbin` to include the bytes one at a time, but I implemented it this way so that I could add more logic later on like combining multiple source bytes into one included byte.

vhelin, 2022-03-04:
I have to say that I have no clue as I didn't implement the -M feature. I can try to take a look this weekend, though...

maxim-zhao, 2022-03-04:
Similarly I found that in -M mode, sometimes I could produce unresolvable calculations. I worked around by passing a define to the -M invocation and some `.ifdef` logic to satisfy the parser in this mode.

I can see in the code that `.fopen` has no code to support makefile mode at all...

rondnelson99, 2022-03-04:
Thanks so much! I can see how widening support for someone else's feature could be difficult. It's a fairy easy issue to work around by adding it as a dependency manually, so don't worry about it *too* much.

maxim-zhao, 2022-03-05:
You could also add something like this:
```
.MACRO incv2bpp ARGS FILE WIDTH ;include a 2bpp row-major image, but convert to column-major
.ifdef MAKEFILE_MODE
.incbin FILE ; hack to make makefile generation work
.else
; existing code
.endif
.endm
```
and pass -DMAKEFILE_MODE when generating the makefile. This would avoid the need to maintain the makefile if you change the source, which the goal of makefile generation anyway.

vhelin, 2022-03-06:
I hope it works now. I'm not 100% sure that what I did is enough (did I understand the feature completely?), but hopefully -M works now with missing files in .FOPEN...

rondnelson99, 2022-03-07:
It seems to work great with the master version. Thank you so much!

vhelin, 2022-03-07:
No problemo, glad I could help you out and make WLA DX a little bit better... :)

closed 2022-03-07 by vhelin