vhelin/wla-dx #493: 6502 Undocumented Opcodes Support

rondnelson99, 2022-03-07:
The NMOS 6502 has many opcodes that appear as a side-effect of how instructions are decoded. These "undocumented", "unofficial" or "illegal" opcodes are detailed [here](https://www.masswerk.at/6502/6502_instruction_set.html#illegals), and they can actually be very useful. I think it would be good for WLA to support these, or at least the ones that are stable. I know that ca65 implements these as a separate "cpu" called [6502X](https://www.cc65.org/doc/ca65-4.html#ss4.3) so that they can be easily disabled if undesired. 

My C skills are not up to snuff for me to implement this, but I'd be happy to do the somewhat tedious step of filling in the opcode table in the source code.

vhelin, 2022-03-07:
Sounds like a plan, I want to add those "illegal" opcodes personally. :)

rondnelson99, 2022-03-07:
Okay, great! First, can you help me understand the format of the opcode table? From the 6502 table:
```
  { "AND #x", 0x29, 4, 0 },
  { "AND (x,X)", 0x21, 4, 0 },
  { "AND (x),Y", 0x31, 4, 0 },
  { "AND x,X", 0x35, 4, 1 },
  { "AND ?,X", 0x3D, 2, 0 },
  { "AND ?,Y", 0x39, 2, 0 },
  { "AND x", 0x25, 4, 1 },
  { "AND ?", 0x2D, 2, 0 },
  { "AND.B #x", 0x29, 4, 0 },
  { "AND.B (x,X)", 0x21, 4, 0 },
  { "AND.B (x),Y", 0x31, 4, 0 },
  { "AND.B x,X", 0x35, 4, 0 },
  { "AND.W ?,X", 0x3D, 2, 0 },
  { "AND.W ?,Y", 0x39, 2, 0 },
  { "AND.B x", 0x25, 4, 0 },
  { "AND.W ?", 0x2D, 2, 0 },
```
I see that `x` represents an 8-bit address, `?` represents a 16-bit address, and `#x` represents an 8-bit immediate. What are the last two fields of these structures? 

Do you think it would be most appropriate to use *all* illegal opcodes or only the stable ones?

I also see that there is a discrepancy between the mnemonics implemented by ca65 and those described in the masswerk.at document. The masswerk.at document doc calls the `(A AND X) - oper -> X` instruction `SBX`, but ca65 implements it as `AXS`. Which should be used here? Maybe someone more knowledgeable about 6502 could comment on which is more common.

vhelin, 2022-03-07:
> I see that x represents an 8-bit address, ? represents a 16-bit address, and #x represents an 8-bit immediate. What are the last two fields of these structures?

The 1st of the last two fields is the ID of the parser in decode_6502.c, the 2nd one is used to signal if lexically the same but with a 16-bit argument opcode follows on the list (more info for the parser in decode_6502.c).

> Do you think it would be most appropriate to use all illegal opcodes or only the stable ones?

All is better, I don't want to choose for the user which ones he/she can use...

> SBX/AXS

I'd say add them both

rondnelson99, 2022-03-07:
Okay, makes sense. It seems like for that second-to-last field, 
0 is implied operand
1 is unused, but same as 5?
2 is 16-bit address
3 is only for `BRK`. I'm not sure how it works but I don't think I need to worry about it.
4 is for 8-bit address is immediate data
5 is for branch offset

Will this come as a new "cpu" like with ca65 or will it be an update to the existing 6502 mode?

vhelin, 2022-03-07:
I'm thinking that it should be just an update to the current 6502 support, no need to make it separate.

vhelin, 2022-03-07:
Do note that the table must be in alphabetical order (first letter only) to work.

closed 2022-06-24 by vhelin