vhelin/wla-dx #496: 65816 - bug with negatives values ?

RetroAntho, 2022-03-18:
Hi,

We updated PVSneslib tool (which use wla 65816) from this commit:
https://github.com/vhelin/wla-dx/tree/6a775a5be7aa59b5588c6eba2b01469217d2bf91
to the last one:
https://github.com/vhelin/wla-dx/tree/663ff0fbfd4a8239df7b209f1189caca8d0b5ddc

but some pieces of code does not work anymore.

It seems to comes from the negatives values (and -1 to be more precise).

This c line code :
```
typedef struct { s16 x, y; }Vec2;
const Vec2 dir[4] = {
	{-2, -1}, {-1, -2}, {1, -2}, {2, -1}
};
```

is converted in asm to :

```
.SECTION "s2c4x.data" APPENDTO "glob.data"
.db $fe,$ff,$ff,$ff,$ff,$ff,$fe,$ff,$1,$0,$fe,$ff,$2,$0,$ff,$ff
.ENDS
```

but when it is converted to binary, it becomes :
0xFFFE for -2 value (seems to be correct)
0x00FF for -1 (= 255) and the bug seems to be here.

As i am not good in asm, i did not understand how is managed the negative value.
Just for my personal culture, how it is done ?

Do you confirm that the behaviour is not the good one ? Is it a bug or how i could solve it ?

Thank you in advance

vhelin, 2022-03-18:
It seems that this is correct

```
.db $fe,$ff,$ff,$ff,$ff,$ff,$fe,$ff,$1,$0,$fe,$ff,$2,$0,$ff,$ff
```

Also I made a test:

```
        .section "extrabits" appendto "Section1"
        .db "07>"                                                       ; @BT TEST-07 07 START
        .db $fe,$ff,$ff,$ff,$ff,$ff,$fe,$ff,$1,$0,$fe,$ff,$2,$0,$ff,$ff ; @BT FE FF FF FF FF FF FE FF 01 00 FE FF 02 00 FF FF
        .db "<07"                                                       ; @BT END
        .ends
```

... which assembled correctly.

Uh, I don't see a problem here... ?

vhelin, 2022-03-18:
Are you sure the problem is not somewhere else?

RetroAntho, 2022-03-20:
No, i cannot guarantee that the issue is not somewhere else because of my poor asm skills, but I finally tested a lot of commits and i found where the issue appears :

With this commit, the code was still functional :
From 21 Jul 2021 : "Added a missing next_line() to the previous commit "
https://github.com/vhelin/wla-dx/commit/ab5593473f62094f1d2ee99db426050952f3c5d2

with the one just after, it does not work anymore :
From  23 Jul 2021 " The assembler is now able to solve A-B where A and B are labels in the same section. Should fix GitHub issues #15 and #185."
https://github.com/vhelin/wla-dx/commit/3f5dd1cd020f484da308f55e9e711d9201d242a8

Here is the result on the sample with the 2 commits :
[breakout.zip](https://github.com/vhelin/wla-dx/files/8312040/breakout.zip)

I checked the change and i guess that i need to update something but it is not clear for now.
I am trying to investigate in generated asm but i will be happy if you have any information to solve it :)

vhelin, 2022-03-20:
I could try to help, but I need the assembly source code, not the final binaries... :)

vhelin, 2022-03-20:
Also the commit https://github.com/vhelin/wla-dx/commit/3f5dd1cd020f484da308f55e9e711d9201d242a8 is broken in regards to A-B solver, and that was fixed later... Try going backwards from the latest commit to pinpoint the issue.

vhelin, 2022-03-20:
If you can create a minimal case that shows the error, that'd be the best. With all assembly and C source code included.

RetroAntho, 2022-03-21:
> I could try to help, but I need the assembly source code, not the final binaries... :)

I hoped the .sym files could help to understand something :)

> Also the commit [3f5dd1c](https://github.com/vhelin/wla-dx/commit/3f5dd1cd020f484da308f55e9e711d9201d242a8) is broken in regards to A-B solver, and that was fixed later... Try going backwards from the latest commit to pinpoint the issue, if possible.

Not sure to understand because i found the last commit which work for me (https://github.com/vhelin/wla-dx/commit/ab5593473f62094f1d2ee99db426050952f3c5d2) and i tried the commit of 12th March (just before the last one) and the issue was still here.

I need to understand exactly where the issue comes (from PVsneslib asm code or directly from this sample which is in C then translated to asm) to  provide you a minimal code.
It will take me more time to provide you better informations. Do you work on Windows ? If yes, i could provide directly binaries to reduce the number of dependencies if you want to build the sample yourself.
I keep you informed as soon as possible

vhelin, 2022-03-21:
I mean https://github.com/vhelin/wla-dx/commit/3f5dd1cd020f484da308f55e9e711d9201d242a8 doesn't produce you working results, but that commit was broken. It's possible that a later commit works and then it's again broken with the latest commit...

We could assume that the new feature in https://github.com/vhelin/wla-dx/commit/3f5dd1cd020f484da308f55e9e711d9201d242a8 breaks PVSneslib in which case there still is a problem in WLA-65816 calculating A-B in the asm source file where A and B are labels... I could try to produce more tests that test just that feature.

I work mainly on Windows, but I also have a Linux machine at home.

I also hope we can fix this issue asap!

vhelin, 2022-03-21:
Also, don't stress too much about making the most minimal case that exhibits the bug. If the test is relatively small and you can pinpoint where the problem happens I can always try to check it out. I'll let you know if the test case is too big/complex for me... :)

RetroAntho, 2022-03-25:
Hi,

After some tests, i see that the behaviour is different depending of the emulator used and i will consider no$sns as the standard.
I reduced the code and i am still not able to see where the issue comes from but i guess it comes from the init map loop.

Result i should have :
[image]

Current result :
[image]

I attached [here](https://github.com/vhelin/wla-dx/files/8351984/pvsneslib.zip) the version for windows with all dependencies (binaries), it works with msys and python need to be installed too
To use it, you just need to define PVSNESLIB_HOME environment variable pointing to the pvsneslib directory (the format need to be in unix style, for example /c/pvsneslib if you extracted it in C:\pvsneslib)

wla binaries are under devkitsnes folder and i built 2 different versions (the last which work with pvsneslib and the last commit from github which does not work anymore) to ease the tests. If you copy/paste it to bin folder, you will need to rebuild pvsneslib (by executing make in pvsneslib folder) before the sample.

The c code is provided and the result in asm is also avaiable in the same directory. When the c file is translated to .asm, you will have a copy named xxx.asm.backup

If i can provide more things or do other tests, i will be happy to help!

Just a small remark when you execute wla-65816.exe or wlalink.exe without any parameters, the date interval printed in the console is not updated :
"Programmed by Ville Helin in 1998-2008"

RetroAntho, 2022-03-26:
I did more tests and i confirm that it comes from this loop :
```
    // Init map with all bricks
    b=0;
    for(j = 0; j < 10; j++) { // lines
        for(i = 0; i < 4; i += 2) { // columns
            a = blocks[b++];

            if (a < 8) {
                c = (j << 5) + i;
                blockcount++;
                blockmap[0x62 + c] = 13 + (a << 10); // left part of brick
                blockmap[0x63 + c] = 14 + (a << 10); // right part
            }
        }
    }
```

I reduced max value of the variable "i".
The loop get value from the map which has been copied into blocks :
```
const u8 map[0x64] = {                              // default wall of bricks
  7, 8, 8, 8, 8, 8, 8, 8, 8, 7,
  8, 7, 8, 7, 8, 8, 7, 8, 7, 8,
  8, 8, 7, 8, 7, 7, 8, 7, 8, 8,
  8, 8, 8, 1, 3, 3, 1, 8, 8, 8,
  8, 0, 4, 8, 8, 8, 8, 4, 0, 8,
  8, 0, 8, 8, 5, 5, 8, 8, 0, 8,
  8, 0, 4, 8, 8, 8, 8, 4, 0, 8,
  8, 8, 8, 1, 3, 3, 1, 8, 8, 8,
  8, 8, 6, 8, 6, 6, 8, 6, 8, 8,
  7, 7, 7, 7, 8, 8, 7, 7, 7, 7,
};

```
7 should be white, 0 = green, etc ...
To be sure that the issue is not elsewhere, i checked with winmerge the content of pvsneslib files and the breakout.c.asm generated with the 2 versions of WLA, it is the same.

vhelin, 2022-03-26:
Thanks for the infos, I'll try to investigate this issue further tomorrow!

vhelin, 2022-03-27:
It seems that I cannot compile the game

[image]

... but I'll try to look for clues in the asm already inside the archive

vhelin, 2022-03-27:
Ok, is there any way to make the compiler to include the C source code lines inside the compiled .asm file? Currently there is no connection between the .asm and the .c...

vhelin, 2022-03-27:
I see this is done before reading the bytes from blocks[] (map -> blocks)

    memcpy((u8 *) &blocks, &map, 0x64);

Is that memcpy() working as it should? If mean if the data in blocks[] after that is not the same as in map[], then the problem is in memcpy()...

vhelin, 2022-03-27:
Ok after making a local copy of the snes headers and editing their paths I was able to compile, but it ends with an error

[image]

What is that EmptyHandler?

vhelin, 2022-03-27:
Ok, after modifying the Makefile I was able to succesfully build the example. Now the hunt begins. :)

vhelin, 2022-03-27:
Got it to work by doing one simple thing

[image]

I moved map[] inside main().

This means, possibly, that the way the compiler deals with global const arrays is not probably working 100% for one reason or another.

vhelin, 2022-03-27:
I'll try to investigate why that it...

vhelin, 2022-03-27:
It seems that when map[] is a global array then we get this

[image]

... and when map[] is a local array to main() we get this (pretty inefficient)

[image]

I assume that the appending of map[]'s data to glob.data doesn't work or the appended data is not somehow used. What we are are seeing in the case of the broken ROM file is possible just uninitialized memory?

vhelin, 2022-03-27:
Could the issue be something like this:

You have a calculation in some asm file where you calculate "SECTIONEND_glob.data - SECTIONSTART_glob.data"

The commit that broke pvsneslib makes it possible calculate that calculation in the assembler. But if the compiler assumed, like it worked before, that only WLALINK is able to calculate that later, when all APPENDTO sections (into e.g., glob.data) are processed and sections are merged...

vhelin, 2022-03-27:
crt0_snes.asm has this

[image]

You have this in the same source file, but lexically before

[image]

If you move those two sections to the very end of the source file, then WLA cannot calculate SECTIONEND_glob.data-SECTIONSTART_glob.data and it should be delayed for WLALINK to calculate, like before... ?

vhelin, 2022-03-27:
Or just put those sections into another source file, that should work as well, I think. :) 

Anyway, many people have requested that WLA could do calculations like that so that the results could be used with directives that need immediate values.

What I think I can do is to add a switch that'll disable WLA's ability to calculate A-B so it'll work just like before and you don't have to change a thing. :)

vhelin, 2022-03-27:
What if you now try the latest sources and give WLA -d when it assembles crt0_snes.obj?

I rebuild that crt0_snes.obj with -d here on my computer and got this:

[image]

:) 

Now that I return to your initial issue report here it's pretty obvious that disabling the new feature that was added in a commit that broke pvsneslib was kinda one way of fixing this and pretty simple, but yeah, it was a good exercise... :)

RetroAntho, 2022-03-27:
Thanks a lot for your investigations and explanations.
I saw in your first pictures that you use cygwin, i think it could explain some issues you got to build sources. (did you created the PVSNESLIB_HOME env variable?)
In the future, we will probably improve the compatibility to works better with more tools.
I confirm that it works like before when using the -d flag. But i do not know if this functionality (the one which produce the bug) could be usefull for the future of pvsneslib.
Just to be sure, i would like to do some tests as you said, by moving these sections to the end of the file (or outside) :
```
.RAMSECTION "globram.data" BANK $7f SLOT 3 KEEP
.ENDS
.SECTION "glob.data" SUPERFREE KEEP
.ENDS
```
I remember that you alreay created a "PRIORITY" instructions for a previous bug in pvsneslib (few years ago), maybe i could also set PRIORITY to 99 for each section to be sure that it will be the last step done (if i understand correctly the cause)

I do not know if you saw my comment about the period displayed of wla's development when executing wla without parameters : 
"Programmed by Ville Helin in 1998-2008"
In case of you want to update it :)

vhelin, 2022-03-27:
PRIORITY doesn't help in this case, the problem is that WLA can now calculate SECTIONEND_glob.data-SECTIONSTART_glob.data if those labels are defined before the calculation in the same asm source file.

About "Programmed by Ville Helin in 1998-2008", check the two other lines after that... :) They add more to WLA's history...

vhelin, 2022-03-27:
>  (did you created the PVSNESLIB_HOME env variable?)

Yes I did...

RetroAntho, 2022-03-27:
ok, i confirm PRIORITY 99 does not work but moving the 2 sections to the end of file solve the issue too.

I guess it will works if i create a file specally for those 2 sections but this solutions seems me less good.
I thought that the file order in linkfile could have an impact too and i tried but it is not the case.

Both solutions are working but i do not know which is the best one. If it was you, which one you will choose ?

> About "Programmed by Ville Helin in 1998-2008", check the two other lines after that... :) They add more to WLA's history...

Yes i saw the date and version number but i found strange to have a date in the past while you still working on it.

vhelin, 2022-03-27:
> Both solutions are working but i do not know which is the best one. If you were me, which one you will choose ?

Probably I would use the -d switch, and add a comment about it into the beginning of the source file (why we need to use it). This might prevent similar problems in the future if more similar sections (unlikely?) are added to the file...

If you choose to move the sections to the end of the source file, please add a comment there as well that tells the looker why that is done as it is pretty important.

vhelin, 2022-03-27:
> Yes i saw the date and version number but i found strange to have a date in the past while you still working on it.

That line just tells that I was the sole author of WLA DX during those years, after those years other people have added code to WLA DX asl well...

closed 2022-03-27 by vhelin

RetroAntho, 2022-03-27:
ok, it is clear and i will follow your advices. Thank again !