vhelin/wla-dx #498: Add a feature that'll disable writing (GB) ROM header bytes

vhelin, 2022-03-25:
Perhaps make it WLA-GB only or make it work somehow in other versions of WLA as well...

ToriningenGames, 2022-03-25:
I assembled a minimal file:
```
.MEMORYMAP
SLOTSIZE $8000
DEFAULTSLOT 0
SLOT 0 $0000
.ENDME

.ROMBANKMAP
BANKSTOTAL 1
BANKSIZE $8000
BANKS 1
.ENDRO

.BANK 0 SLOT 0
.ORG 0
.db 0
```
Other than the defined byte, WLA-GB filled in $0148 (ROM size) with $00. It issued a warning about the number of banks when doing so. Using a standard GB ROM layout:
```
.MEMORYMAP
SLOTSIZE $4000
DEFAULTSLOT 0
SLOT 0 $0000
SLOT 1 $4000
.ENDME

.ROMBANKMAP
BANKSTOTAL 2
BANKSIZE $4000
BANKS 2
.ENDRO

.BANK 0 SLOT 0
.ORG 0
.db 0
```
$0148 is also filled in. Once more, using a silly map:
```
.MEMORYMAP
SLOTSIZE $333666
DEFAULTSLOT 0
SLOT 0 $333000
SLOT 1 $666
.ENDME

.ROMBANKMAP
BANKSTOTAL 1
BANKSIZE $9
BANKS 1
.ENDRO

.BANK 0 SLOT 0
.ORG 0
.db 0
``` 
Just like the first map, alongside another complaint the bank is smaller than the slot (expected and good).
Silly bank count? No, WLA-GB is limited to 512 banks total. This limitation is not present in WLA-6502.
From this, it seems the only thing pertaining to the issue is how WLA-GB treats `.ROMBANKMAP`, specifically the bank count.

ToriningenGames, 2022-03-25:
If we added a `.ROMBANKCOUNT` directive, it'd be a breaking change. But, it would put WLA-GB in line with the other assemblers.
Thankfully, the Game Boy doesn't actually care about the value. So, running assembled games in emulators (common) or GB programs that read those bytes themselves (rare I hope) would be affected.

vhelin, 2022-03-26:
"... WLA-GB is limited to 512 banks total..." This limitation comes from the fact that AFAIK there is no suitable value in $0148 for bank count larger than 512. But if the user specified 1024 banks I could always write 8 (512 banks) there..

> Thankfully, the Game Boy doesn't actually care about the value. So, running assembled games in emulators (common) or GB programs that read those bytes themselves (rare I hope) would be affected.

Has it been thested that the real physical Gameboy doesn't care about the value? Emulators can do whatever they want, but being compatible with real Gameboy is still preferred...

vhelin, 2022-03-26:
Use now .ROMSIZE with WLA-GB to write the ROM size indicator value at $0148. There was already .RAMSIZE that did the same for RAM size at $0149 so using .ROMSIZE instead of .GBROMBANKCOUNT felt logical. Though .RAMSIZE and .ROMSIZE are not the best of directive names, I admit...

cr1901, 2022-03-26:
`.GBROMBANKCOUNT` and `.GBRAMBANKCOUNT` :)?

ToriningenGames, 2022-03-26:
I ran some official bootloaders through an emulator (I own a device with each bootloader), and found that neither the original Game Boy nor the Game Boy Color read $0148 except for calculating checksum. The GBA uses the GBC bootloader. That just leaves the Super Game Boy v1 and v2, and the Pocket Gameboy (apparently, according to Pan Docs, it uses a different bootloader? I don't have one to check). Although I have an SGB, I can't really dig into it for testing that. According to the Pan Docs, neither SGB cares about $0148.

Conventional wisdom holds the data in the header was for Nintendo's benefit, in manufacturing or handling customer complaints. That isn't conclusive. If somebody had each Game Boy compatible... thingy... we could likely plug in a bus analyzer to see if it reads $0148 more than once. I don't have one, though. Should probably work on that.

vhelin, 2022-03-26:
> `.GBROMBANKCOUNT` and `.GBRAMBANKCOUNT` :)?

I think that if we add aliases to those directives we should add aliases to many others at the same time as there are others that'd could use that GB prefix... :)

vhelin, 2022-03-27:
@ToriningenGames Anyway, does the .ROMSIZE directive solve this issue, what do you think?

ToriningenGames, 2022-03-28:
My tests come out fine. I didn't verify whether .ROMSIZE operates correctly; I assume it works as well as WLA-GB worked before. But, not using it allows assembly without GB restrictions or extra added bytes. So, good for me.
Thanks!